ethercrab-rs/ethercrab #12: See if we can migrate from `packed_struct` to `packing` crate

jamwaffles, 2023-01-20:
`packed_struct` has been a bit of a pain to use. It's likely I'm just holding it wrong, but let's see what [`packing`](https://crates.io/crates/packing) can do instead. Will likely need to be coupled with [`num_enum`](https://crates.io/crates/num_enum).

jimy-byerley, 2023-05-05:
I'm coming late, but I tried `packing` a bit:
- it has a `packing::Packed` trait very similar to `packed_struct::PackedStruct`
- very nice syntax, (less string quotes on numeric args :innocent:, less args as well) I could define a `ControlWord` struct just like this:
```rust
#[derive(Packed, Copy, Clone, Default, Debug)]
#[packed(little_endian, lsb0)]
pub struct ControlWord {
	// pkd(start_bit, end_bit, start_byte, end_byte)   is zero-based and must have start==end to signify length 1 item
	
	#[pkd(7,7,0,0)]  pub reset_fault: bool,
	#[pkd(3,3,0,0)]  pub enable_operation: bool,
	#[pkd(2,2,0,0)]  pub quick_stop: bool,
	#[pkd(1,1,0,0)]  pub enable_voltage: bool,
	#[pkd(0,0,0,0)]  pub switch_on: bool,
	#[pkd(0,0,1,1)]  pub halt: bool,
}
```

This is a bit similar to `packed_struct` but somehow I find `packing` cleaner

jamwaffles, 2023-05-18:
That example looks pretty nice. I just found [bilge](https://hecatia-elegua.github.io/blog/no-more-bit-fiddling/) which could be even nicer, although [big endian isn't supported yet](https://github.com/hecatia-elegua/bilge/issues/7). EtherCAT is LE but IIRC I've used BE in a couple of places (although don't recall why).

jimy-byerley, 2023-05-18:
So there is two limitations:
- It only supports little endian
- The size of structs is limited to 128 bits

I guess they both might evolve one day ..
Anyway, `bilge` looks awesome ! I will git it a try

Were you not using big endian in some places just for byte-level debug readability ? :thinking:

hecatia-elegua, 2023-05-24:
Apparently I can't choose to get notified when someone links to one of bilge's issues? lol
Anyways, I just saw this.

Please tell me more about where and how you need big endian in the bilge issue. Someone else said they'd handle endianness at the IO layer, so depending on what you're doing here, maybe `SomeBitfield::from(u32::from_be(hardware_value))`.
Same for the size usecase - can you split up the struct? Why not?

jimy-byerley, 2023-05-24:
Hi @hecatia-elegua ! It seems we invoked you here

- about endianness at IO layer, I think it could be a solution, however it prevents any mix of big and small endian in the same struct or in nested structs ...
- about struct size, of course it is almost always possible to split up the struct, but still more readable to have all the fields at the same struct level when grouping has no specific meaning in itself ;)   it is a debate about encoding versus expressing ...

hecatia-elegua, 2023-05-24:
Right, mixed-endian inside a bitfield is unsupported. Would love to know more about where/why that's useful, but I'll support it sooner or later.
I get the point about readability, but array code would worsen maintenance and ergonomics, so it's a kind-of trade-off. Maybe it would be possible to split the struct inside macro code. Can you link me the code where this is needed?

jimy-byerley, 2023-05-25:
The following is a typical struct that would need bit fields in the middle of a regular struct

The following is a frame from the ethercat specs

| Data field | Data type | Value/description |
|-----------|------------|-------------------|
| CMD | Unsigned8 | 0x01 (command APRD)
| IDX | Unsigned8 | Index
| ADP | WORD | Auto increment address
| ADO | WORD | Physical memory or register address
| LEN | Unsigned11 | Length of the DATA data field
| reserved | Unsigned3 | 0x00
| C | Unsigned1 | Circulating frame  0: Frame is not circulating, 1: Frame has circulated once
| NEXT | Unsigned1 | 0: last EtherCAT PDU in EtherCAT frame  1: EtherCAT PDU in EtherCAT frame follows
| IRQ | WORD | External event
| DATA | OctetString LEN | Data, structure as specified in 5.6, Clause 6 or by DLS-user
| WKC | WORD | Working counter

And it is [implemented here in ethercrab](https://github.com/ethercrab-rs/ethercrab/blob/9b55183df07691c00465ebe7e253a4eb71c27b0a/src/pdu_loop/frame_element/sendable_frame.rs#L68)

With such long struct, we need
- long memory layout (I think that one has 10 bytes before data, so it does fit in 128 bits but some other frames might be bigger)
- little endian on every fields even aligned (thus it is good to rely on a packing library like bilge)
- precise bit layout of some fields
- since the specs mention this struct as one struct, it is better for readability to implement it as one struct as well

This is as an example :)

hecatia-elegua, 2023-05-25:
Thank you for the thorough example, I'll have to take some time to look at this further!

Just one thought about the last point:
I think there are worse problems with specs than mapping all registers exactly to one struct. I would even say readability is _better_ if you split it up: A spec will always give one register one table, since readability doesn't really matter there, it's just about regularity. That could mean you end up with a table containing 32 different fields or more, which is often a bit bloated in code. There will probably be some parts of these fields which are less accessed, or used for slightly different purposes, which would be better off in different bitfield structs and then combined in a parent struct.
Maybe that would be possible here?

Wait, you're doing this already anyways: https://github.com/ethercrab-rs/ethercrab/blob/9b55183df07691c00465ebe7e253a4eb71c27b0a/src/pdu_loop/pdu_flags.rs#L5

jimy-byerley, 2023-05-25:
I agree that in general, specs often do not group values by meaning and then introducing better grouping of data can improve readability or efficiency. I would event say it is only a matter of design. However in the latter example, grouping `LEN`, `C`, `NEXT` has really no meaning as they are used for completely different purpose. grouping here is only to save bytes, so readablity is better when not grouped.

Well of course it is always possible to group them anyway :sweat_smile: . I'm  just arguing that in such case, grouping them is less convenient. And bit packing libraries are all about convenience isn't it ?
I don't know why [they are actually packed separately](https://github.com/ethercrab-rs/ethercrab/blob/9b55183df07691c00465ebe7e253a4eb71c27b0a/src/pdu_loop/pdu_flags.rs#L5), maybe it is because when writing it all with bitwise ops it felt simpler that way @jamwaffles  ?

PR #142 (2024-01-07): Use `ethercrab-wire` derives, remove `PduRead` and `PduData`
Removes the two aforementioned traits and replaces them with `EtherCrabWireRead` and `EtherCrabWireReadWrite` respectively.

This PR allows the removal of `packed_struct` and `num_enum`, so closes #12 :tada:.

(no shade on either of these crates - they're both great - I just want/need more control over data in EtherCrab)

This PR is a split of the unwieldy #139
closed 2024-01-07 by jamwaffles