ethercrab-rs/ethercrab #21: example `ec400` stalling with slave state not switching to OP

jimy-byerley, 2023-03-14:
Hello
This may be the first of a serie of stupid interogations for setting up a master using CoE

I'm trying to start a cyclic communication with PDO read/write on an omron servodrive (R88D-1SN04H-ECT), based on `example/ec400.rs`. While starting, the program stalls in the loop waiting for `servo.tick()`. The slave never switches to OP and after ~2min, the program panics with the following log:

```rust
ydejonghe@UBUNTU:~/ethercat/ethercrab$ sudo RUST_LOG=debug target/debug/examples/ec400
[2023-03-14T13:55:29Z INFO  ec400] Starting SDO demo...
[2023-03-14T13:55:29Z DEBUG ethercrab::slave] Slave 0x1000 name R88D-1SN04H-ECT
[2023-03-14T13:55:29Z DEBUG ethercrab::client] Configuring topology/distributed clocks
[2023-03-14T13:55:29Z DEBUG ethercrab::dc] Slave 0x1000 R88D-1SN04H-ECT has DC
[2023-03-14T13:55:29Z DEBUG ethercrab::dc] --> Times 39335733 (2139694680) 2179030413 (0) 0 (1819436374) 1819436374
[2023-03-14T13:55:29Z DEBUG ethercrab::dc] --> Propagation time None ns, child delay 0 ns
[2023-03-14T13:55:29Z DEBUG ethercrab::dc] Distributed clock config complete
[2023-03-14T13:55:29Z DEBUG ethercrab::dc] Performing static drift compensation using slave 0x1000 R88D-1SN04H-ECT as reference. This can take some time...
[2023-03-14T13:55:31Z DEBUG ethercrab::dc] Static drift compensation complete
[2023-03-14T13:55:31Z DEBUG ethercrab::slave_group::configurator] Going to configure group, starting PDI offset 0x000000
[2023-03-14T13:55:31Z DEBUG ethercrab::eeprom] Discovered sync managers:
    [
        SyncManager {
            start_addr: 0x1800,
            length: 0x0100,
            control: Control {
                operation_mode: Mailbox,
                direction: MasterWrite,
                ecat_event_enable: false,
                dls_user_event_enable: true,
                watchdog_enable: false,
            },
            enable: ENABLE,
            usage_type: MailboxWrite,
        },
        SyncManager {
            start_addr: 0x1c00,
            length: 0x0100,
            control: Control {
                operation_mode: Mailbox,
                direction: MasterRead,
                ecat_event_enable: false,
                dls_user_event_enable: true,
                watchdog_enable: false,
            },
            enable: ENABLE,
            usage_type: MailboxRead,
        },
        SyncManager {
            start_addr: 0x1100,
            length: 0x000c,
            control: Control {
                operation_mode: Normal,
                direction: MasterWrite,
                ecat_event_enable: false,
                dls_user_event_enable: true,
                watchdog_enable: true,
            },
            enable: ENABLE,
            usage_type: ProcessDataWrite,
        },
        SyncManager {
            start_addr: 0x1200,
            length: 0x001c,
            control: Control {
                operation_mode: Normal,
                direction: MasterRead,
                ecat_event_enable: false,
                dls_user_event_enable: true,
                watchdog_enable: false,
            },
            enable: ENABLE,
            usage_type: ProcessDataRead,
[... 533 lines cut ...]
            let vel = i32::from_le_bytes(i[6..=9].try_into().unwrap());

            (pos, vel)
        };

        println!(
            "Position: {pos}, velocity: {vel}, status: {status:?} | {:?}",
            o
        );
    }

    Ok(())
}

fn main() -> Result<(), Error> {
    env_logger::init();
    let local_ex = LocalExecutor::new();

    futures_lite::future::block_on(local_ex.run(main_inner(&local_ex))).unwrap();

    Ok(())
}
```

Do you think I am doing something wrong ?

jimy-byerley, 2023-03-14:
I can see on wireshark, that the slave actually received the request for mode OP and answered two successive state changes as expected: SafeOP then OP

Here is the transmission capture: [issue-21.pcapng.zip](https://github.com/ethercrab-rs/ethercrab/files/10970718/issue-21.pcapng.zip)

[image]

jamwaffles, 2023-03-14:
Thanks for the bug report and Wireshark dump. A few things:

- You're still checking for `if slave.name() == "ZeroErr Driver" {`, which means the drive won't be configured correctly to go into OP. I don't see `using it` printed in the logs which confirms this. If you have no other slaves on the network, try removing the `if` and doing the SDO writes unconditionally.
- I noticed the slave name is "R88D-1SN04H-ECT\0\0\0\0\0"` which looks like a bug in either EtherCrab or the drive firmware. Could you please rerun your program with EtherCrab on the `trace` level (e.g. `RUST_LOG=ec400=debug,ethercrab=trace`) and attach the resulting output as a file - it will be very long - so I can debug this name reading issue please?
- The packets you've highlighted in wireshark are EtherCAT's AL (Application Layer) statuses. DS402 has its own status on top of this which, confusingly, uses a lot of the same names. AFAIK DS402 isn't decoded by Wireshark so you'll just get raw `LRW` packet data there. Your drive is going into _EtherCAT_ `OP` which then allows the DS402 state machine to start attempt to put the drive into _DS402_ `OP` state, where it's failing for some reason. You should be seeing `DEBUG ethercrab::ds402] Edge ...` spewing everywhere but I'm not sure why it's hanging.

As mentioned above, a `trace` level log and a new wireshark capture if you've changed anything would be appreciated, e.g. `RUST_LOG=ec400=debug,ethercrab=trace`

My best guess right now is that the drive isn't configured at all as none of the SDOs are written in the init closure. Try removing the `if` that checks for the slave name and see if you get any SDOs written by looking at Wireshark or the log output. Have a look at your drive's manual on correct values for CSV as well - the EC400 examples uses values specific to that drive.

It's late for me here but tomorrow I'll test `master` branch with my EC400 drive to check if there have been any regressions since I ran it last - it's strange that you're only seeing one DS402 tick.

jimy-byerley, 2023-03-15:
Oops I was running the test agains the two types of servodrives, and I copy-pasted the wrong code here it seems. I ran the program again with the good condition: mention `using it` is present this time and the same stall occurs.
(For the coming tests I will remove the name condition to avoid any possible problem with it).

I noticed that strange name issue, not sure if it is a bug in the firmware or not, but `etherlab-rs` doesn't seems to suffer from it. 
Here is the full trace:  [issue-21-2.zip](https://github.com/ethercrab-rs/ethercrab/files/10977433/issue-21-2.zip)

I will check for the drives actual pdo addresses. But shouldn't the drive raise errors if I'm using wrong PDOs ?

jamwaffles, 2023-03-15:
Hey, the string name issue _should_ be fixed in #23 but it would be great if you could confirm if it fixed it for you or not. I don't have access to my test gear right now, but none of them showed the bug anyway...

> But shouldn't the drive raise errors if I'm using wrong PDOs?

I don't think so - I think they just fail to go into DS402 OP if I remember what my drives do correctly.

jimy-byerley, 2023-03-15:
I confirm #22 is fixed in my case :)

jimy-byerley, 2023-03-15:
> Have a look at your drive's manual on correct values for CSV as well - the EC400 examples uses values specific to that drive.

You were right, I did paid attention to that my PDO addresses were right (they are the same for omron drives and leadshine drives), but forgot to check the SDO selecting the PDO to receive (those are differents between omron and leadshine)
I will try with the correct ones

jamwaffles, 2023-03-15:
I just ran the `ec400` demo on my servo drive and it still works, so no regressions 🎉 

You might find it useful to look for `Discovered PDOs` in the logs - the list of PDOs and their values are listed below that message. The manual is probably a better source of truth, but it might be useful for cross checking.

jimy-byerley, 2023-03-17:
Well my tests gets weirds ...
I spent the last day checking the SDO and PDO addresses and those used in `ec400.rs` are the same for leadshine and omron (how cumbersome it is with the XML dumb syntax !).
Maybe what was wrong in my test was that I commented the writing of SDOs `0x1c12`  and `0x1c13` however they are assigning the PDOs to the sync managers. So I guess that was pretty regular that the servodrive was not sending frames.
What is weird is that when I uncomment the writing of these two registers, It gets refused by the servodrive:
```rust
slave.write_sdo(0x1c12, SubIndex::Index(0), 0u8).await?;
slave.write_sdo(0x1c12, SubIndex::Index(1), 0x1600).await?;
slave.write_sdo(0x1c12, SubIndex::Index(0), 1u8).await?;
```
```rust
thread 'main' panicked at 'Init: Mailbox(Aborted { code: General, address: 7186, sub_index: 1 })', examples/ec400.rs:129:10
// 7186 = 0x1c12
```

jimy-byerley, 2023-03-17:
> You might find it useful to look for Discovered PDOs in the logs - the list of PDOs and their values are listed below that message. The manual is probably a better source of truth, but it might be useful for cross checking.

I think that will be of great help in further steps of my project, but yet I'm struggling even at triggering the PDOs :sweat:

closed 2023-03-17 by jimy-byerley

jamwaffles, 2023-03-17:
Great to hear! SDO issues are pretty hard to debug so well done figure it out :). I'm not sure it's possible but if it is, I'll look at having an error to report incorrect data types because this is a pretty mean gotcha in the current API.

jimy-byerley, 2023-03-17:
The SDO abort code helped me actually
In the wireshark log, there was a field `Abort code: 0x06070010` . Refering to the canopen communication profile, I found
`0607 0010 h     Data type does not match, length of service parameter does not match`

Since these error codes are part of the canopen specifications, it could be great to mention the received error code and eventually its string message in the error returned by `write_sdo()`:
```rust
Mailbox(Aborted {
   code: General, address: 7186, sub_index: 1 
   errcode: ...,  // maybe even implement Display so it can be printed with the error message provided by the canopen specs
   })
```