ethercrab-rs/ethercrab #22: EEPROM strings contain trailing null bytes

jamwaffles, 2023-03-15:
Issue #21 raised a bug where strings read from the slave device contain trailing null bytes (`\0`). These should be trimmed from the returned string, otherwise we get behaviour like a slave name of `"R88D-1SN04H-ECT\0\0\0\0\0"`.

Presumably this slave device assumes a language with null-terminated strings is reading the EEPROM, so is a bit loose about what it returns after the string data. In this specific case, the slave device reports a length of 20 bytes for the string, where `"R88D-1SN04H-ECT"` is only 15 bytes long.

closed 2023-03-15 by jamwaffles

jimy-byerley, 2023-03-15:
So this was an issue with the servodrive's firmware, interesting :thinking:

jimy-byerley, 2023-03-15:
I would guess the same, but since the ethercat specs do not mention the names should be read as null-terminated strings, I find Omron tricky to rely on it ;)

jamwaffles, 2023-03-15:
No, even better, EtherCAT SII strings have a length field just before them so really there's no excuse 😁 but this is where the real world meets EtherCrab, and the realworld up till now has been mostly C :(