ethercrab-rs/ethercrab #27: See if it's possible to report a better error for incorrect datatypes

jamwaffles, 2023-03-17:
[For example](https://github.com/ethercrab-rs/ethercrab/issues/21#issuecomment-1474045178),

```rust
// was doing this:  implicitely a usize I guess, implying u64
slave.write_sdo(0x1c12, SubIndex::Index(1), 0x1600).await?;
// instead of this: u16
slave.write_sdo(0x1c12, SubIndex::Index(1), 0x1600_u16).await?;
```

The error is `Init: Mailbox(Aborted { code: General, address: 7186, sub_index: 1 })` which doesn't help the user much. If possible, EtherCrab should tell the user they've used the incorrect length value (or better, incorrect type) for that field. This information _is_ read from the EEPROM during PDO discovery so maybe that could be used.

jimy-byerley, 2023-03-17:
The canopen specifications ask the slave to return an error code with that Abort object, so it is possible to retreive the cause of the failure without actually adding any checking on the ethercrab side.

- A minimal change could be to add this error code in the `Aborted` rust error
- A even better solution could be to implement `Display` for `Aborted` so it can print the specified canopen message matching this error code

jamwaffles, 2023-03-17:
Note to self: Looks like a parsing bug [here](https://github.com/ethercrab-rs/ethercrab/blob/9798ac9778d6bb1173067c59a5f164237b2dc51e/src/slave/mod.rs#L511). I'll reproduce a similar error with my servo drive and check it decodes properly or fix the bug.