ethercrab-rs/ethercrab #40: What about tools to manipulate SDOs and PDOs

jimy-byerley, 2023-04-06:
Hello @jamwaffles 

I was working on a benchmark of `ethercrab` vs `ethercat-rs` as we mentioned, and I've found useful the following piece of API. I didn't fully implemented it due to some difficulties I will explain after.

## API suggestion

### base concepts

The idea is to have pointer-like variables pointing to fields in datagrams, and other to SDOs on slaves. However those would be perfectly safe because they would not be pointers but bare indices with runtime bounds checks.

```rust
/// an instance of this refers to some data in a datagram
struct Field<T: PduData> {
  pub byte: usize,   // start byte offset in the datagram
  pub bit: u8,    // start bit offset in the start byte
  pub bitlen: usize,  // bit length of the value after the start byte
}

/// an instance of this refers to an SDO item or subitem
struct Sdo<T: PduData> {
  pub index: u16,
  pub sub: u8,
}
```

After this there is two alternatives:

1. they shall provide methods to get/set what they are refering to

```rust
impl Field<T> {
  fn get(&self, data: &[u8]) -> T
  fn set(&self, data: &mut [u8], value: T)
}

impl Sdo<T> {
  async fn get(&self, slave: &SlaveRef) -> Result<T> 
  async fn set(&self, slave: &SlaveRef, value: T) -> Result<()>
}
```

2. what they are refering to shall provide methods to get/set them, as if the aformentioned structs were keys in a table

```rust
impl [u8] {
  fn get(&self, key: &Field<T>) -> T
  fn set(&mut self, key: &Field<T>, value: T)
}

impl Slave {
  async fn get(&self, sdo: &Sdo<T>) -> Result<T>
  async fn set(&self, sdo: &Sdo<T>, value: T) -> Result<()>
}
```

### usage example

```rust
/// in a declaration file
let operation_mode = Sdo::<u16> {index: 0x6040, sub: 0};
let position = Sdo::<i32> {index: 0x6064, sub: 0};
/// in user code
operation_mode.set(slave, OperationMode::SynchronousPosition as u16).await?;
let p = position.get(slave);
```

## an now, even more helper structs

In the same spirit, I'm also considering helper structs to configure the PDO mappings

```rust
struct ConfigurablePdo {
  index: u16,  // sdo address
  num: u8, // number of sdo subitems that can be mapped
}

// helper to init a sync manager mapping, compute and increment mapped fields offsets
struct SyncMapping {...}
impl SyncMapping {
  fn push(&mut self, sdo: &ConfigurablePdo) -> PdoMapping
  fn finish(&mut self)
}
// helper to init a pdo mapping, compute and increment mapped fields offsets
struct PdoMapping {...}
impl PdoMapping {
  // produce a Field pointing to where the sdo subitem will be in the received/transmitted datagram, so the user can use it to easily get/set the value in the datagram
  fn push(&mut self, sdo: &Sdo<T>) -> Field<T>
  fn finish(&mut self)
}
```

### usage example

```rust
// in declaration file
let profile = {.. some big nested struct of helpers and Sdo<T> ...}

// in pre-op mode
let mapping = slave.mapping().await?;     // start configuring sync manager mapping
let pdo = mapping.push(&profile.pdo[0]).await?;    // add a PDO and start configure it
current_status = pdo.push(&profile.current.status, sdo_status).await?;   // set next (first) object to status SDO
current_position = pdo.count(&sdo_position).await?;   // assum next (second) object is position SDO
pdo.finish().await?;
mapping.finish().await?;

// in op mode
let (i, o) = slave.io();    // i and o are actually datagrams, respectively &[u8] and &mut [u8]
let status = current_status.get(i);
let pos = current_position.get(i);
target_position.set(o, pos+some_increment);
```

## state of work

I implemented it, but it only works for `SlaveRef`, since the get/set method have to take a slave as argument It cannot be both `SlaveRef` and `GroupSlave`.
By the way, what is the distinction between `SlaveRef` and `GroupSlave` ? They both access SDOs

jimy-byerley, 2023-05-04:
I tested the proposed functions and structs yesterday, and I'm quite satisfied with them. What would you think of a PR ?

jamwaffles, 2023-05-05:
I just merged #47 which is enormous, but buried within it removes `GroupSlave`s entirely. Hopefully a) you don't get too many conflicts and b) it makes your implementation a little easier.

jamwaffles, 2023-05-05:
The PRE-OP configuration idea looks quite nice, and might fit well into something that parses from ESI files.

In terms of reading/writing during OP mode, I was considering an API something like this very pseudo-code-ey idea:
Example 1: Servo drive

```rust
struct Ds402<'a> {
    pdi: SlavePdi<'a>

    // DS402 state machine, whatever else it needs
}

impl Ds402 {
    fn position(&self) -> u32 {
        self.position
    }

    fn set_position(&mut self, target: u32) {
        self.target_pos = target
    }

    fn state_machine_tick(&mut self) {
        // ...
    }
}

loop {
    tick.await;

    group.tx_rx().await;

    // NOTE: Need to figure out how to do `mut` here!
    let mut slave = group.slave::<SlaveRef<'_, Ds402>>(0);

    println!("Current pos: {}", slave.position());

    slave.set_position(slave.position() + 100);
}
```

Example 2: EL2004 quad output

```rust
struct El2004<'a> {
    pdi: SlavePdi<'a>
}

enum Pin {
    Pin1,
    Pin2,
    Pin3,
    Pin4,
}

impl Ds402 {
    fn set_pin(&mut self, pin: Pin, state: bool) {
        // ...
    }
}

loop {
    tick.await;

    group.tx_rx().await;

    // NOTE: Need to figure out how to do `mut` here!
    let mut slave = group.slave::<SlaveRef<'_, El2004>>(0);

    // Turn first pin on
    slave.set_pin(Pin::Pin1, true);
}
```

I like the idea of using typestates for this, even if they are validated only at runtime. I still have questions around that validation, e.g. if I try to select slave `0` as an `El2004` but it is actually a `Ds402` servo drive, that needs to return an error, but how do we check that?

Anyway, what do you think? I'm presenting the above as an idea, not the final solution but I'd be interested to hear your feedback.

jimy-byerley, 2023-05-06:
regarding slave manipulations, I am hesitating about defining a fixed struct to manage a standard slave: Most of the slaves I know do have standard features in common, but often coming with a bunch of manufacturer specific features and behaviors. Think of servodrives: they all respect cia402 definitions, but
- some will have many modes of operations, some will have only few
- some will allow operation mode switch, some will not, some will allow only switching from some modes to some others but not switching back
- same for torque limitations
- some will use manufacturer specific homing procedures, or powering on cycles

So I'm sot sure that it will be easy to write a struct for the features common to every cia402 servo drives for example. The common features set is very small if we consider the poorest drives implementing cia402, so the struct written for it would have to either: implement ony very few functions, or implement a lot of functions for all optional features, and half of those will not be available on most servodrives ...

 Facing this, I am personaly considering a slightly different approach:
 
 - the user *has to* create its own structs, for its specific devices, its specific needs, and the specific feature set of the devices
 - the library *will* provide tools to easily interact with the ethercat devices, independently of their type
 - it does not prevent the library to provide structs to manage specific features, but they *shall* not be too opaque, and *shall* interoperate with other structs managing other features of the same device.

To summarize, I like the idea of *one struct per device feature rather than one struct per device type*. And the user will instantiate these structs according to the actual features of its devices. 

This is the way the `SyncMapping` I propose is working:  it only manage a specific feature of a set of devices (this feature is PDO mapping, concerning complex ethercat slaves), and the user may use this struct or not to configure its device, and SyncMapping is not taking full control over the device.

The `Field<T>` returned by the PDO mappings may help this:  it makes the user able to easily read/write SDO without particular knowledge of their addresses and data types. So I think letting the user interact with the process data  through `Field<T>` instances is fine.

Here are my thoughts :relieved: 

> I like the idea of using typestates for this

I like it typestate too :) as long as it does not make the code much more verbose or does not constrain too much the architecture of the user's program.
I got an idea of typestate coding style to ease the rust's ownership of the ethercat client struct. I will tell you about this when my idea is more mature.

> even if they are validated only at runtime

how are they validated at runtime ?! isn't this only using the compiler's type-checking ?

jamwaffles, 2023-07-14:
Thanks for the discussion, however this isn't a direction I'd like to take EtherCrab in at this time.