ethercrab-rs/ethercrab #41: cannot run `ec400`: panicked at 'Init: WorkingCounter`

jimy-byerley, 2023-04-11:
Hello, it's me again
I have an issue running the latest commit

```
ethercrab$ RUST_LOG=debug sudo -E ./target/release/examples/ec400 eno1
[2023-04-11T10:11:23Z INFO  ec400] Starting EC400 demo...
[2023-04-11T10:11:23Z INFO  ec400] Ensure an EC400 servo drive is the first and only slave
[2023-04-11T10:11:23Z INFO  ec400] Run with RUST_LOG=ethercrab=debug or =trace for debug information
[2023-04-11T10:11:23Z DEBUG ethercrab::slave] Slave 0x1000 name R88D-1SN02H-ECT
[2023-04-11T10:11:23Z DEBUG ethercrab::client] Configuring topology/distributed clocks
[2023-04-11T10:11:23Z DEBUG ethercrab::dc] Slave 0x1000 R88D-1SN02H-ECT has DC
[2023-04-11T10:11:23Z DEBUG ethercrab::dc] --> Times 1955227410 (0) 1717989224 (0) 0 (1819436374) 1819436374
[2023-04-11T10:11:23Z DEBUG ethercrab::dc] --> Propagation time None ns, child delay 0 ns
[2023-04-11T10:11:23Z DEBUG ethercrab::dc] Distributed clock config complete
[2023-04-11T10:11:23Z DEBUG ethercrab::dc] Performing static drift compensation using slave 0x1000 R88D-1SN02H-ECT as reference. This can take some time...
[2023-04-11T10:11:25Z DEBUG ethercrab::dc] Static drift compensation complete
[2023-04-11T10:11:25Z DEBUG ethercrab::slave_group::configurator] Going to configure group, starting PDI offset 0x000000
thread 'main' panicked at 'Init: WorkingCounter { expected: 1, received: 0, context: Some("debug write") }', examples/ec400.rs:118:10
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

It is fine with the previous commits, however I'm having troubles to start the cyclic communication (this is probably a different issue)

jamwaffles, 2023-04-11:
Hey jimy, thanks for the report. I don't know how much time/energy I have to devote to ethercrab at the moment but I'll look at this when I can if you haven't solved it already :)

jimy-byerley, 2023-04-11:
No problem, I just put it here so you have it somewhere ;)

jamwaffles, 2023-04-11:
Much appreciated! It's useful for tacking so please post any more issues you find.

P.s. I've looked at #40 and have taken it in. I still don't have a concrete answer yet but just wanted to let you know I'm not ignoring it :grin:

jamwaffles, 2023-04-13:
Reproduced on my EC400 drive with just the latest commit in `master`. Will do some more investigation and find a fix 👍

jamwaffles, 2023-04-14:
Fixed by #42. Please give it a test and reopen if you still hit the issue. I tested on my EC400 servo drive and it works for me :)

closed 2023-04-14 by jamwaffles

jimy-byerley, 2023-04-14:
It works again, thanks !

PR #42 (2023-04-14): Fix slave storage issues
Closes #41 

This was a regression caused by the introduction of the internal `SlaveStorage` container - it was writing the slave device into the wrong piece of memory, meaning the retrieval of that slave's info would return a struct with all zeroes. This caused the error seen in #41 as it was sending an address of `0x0000` instead of `0x1000` (which the slave device is configured to.
commit 1c48b416 (2023-04-14): Fix slave storage issues (#42)

* Fix slave storage doing the wrong thing

Closes #41