ethercrab-rs/ethercrab #49: error with 3 servodrives `Init: Timeout`

jimy-byerley, 2023-05-09:
Hello hello, that's me again with new problems

I'm trying to run `ec400`  on a chain of 3 servodrives, but I'm facing a timeout error and I don't really know what is its cause since when connecting to 1 or 2 only of these slaves, the problem disappears.
The following backtrace is with the current master branch, but it was also happening with the previous commits.

[dump.txt](https://github.com/ethercrab-rs/ethercrab/files/11428983/dump.txt)

Any idea ?

jimy-byerley, 2023-05-11:
Here are some wireshark dumps: [dump-timeout.zip](https://github.com/ethercrab-rs/ethercrab/files/11449381/dump-timeout.zip)
- with 3 omron servodrives it is working fine
- with 4 omron servodrives it panics with a timeout
- with 3 omron servodrives and 3 zeroerr servodrives it panics with a timeout

The following is the log at level *info*, log at level *debug* is in the archive

```
ydejonghe@UBUNTU:~/ethercat/ethercrab$ cargo build --release --example ec400
   Compiling ethercrab v0.1.0 (/home/ydejonghe/ethercat/ethercrab)
    Finished release [optimized] target(s) in 7.40s
ydejonghe@UBUNTU:~/ethercat/ethercrab$ sudo RUST_LOG=info target/release/examples/ec400 eno1
[2023-05-10T16:02:56Z INFO  ec400] Starting EC400 demo...
[2023-05-10T16:02:56Z INFO  ec400] Ensure an EC400 servo drive is the first and only slave
[2023-05-10T16:02:56Z INFO  ec400] Run with RUST_LOG=ethercrab=debug or =trace for debug information
^C[2023-05-10T16:04:01Z ERROR ethercrab::slave] Mailbox read ready error: Timeout
thread 'main' panicked at 'Init: Timeout', examples/ec400.rs:117:10
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I also tried with different values of `Timeouts`, by multiplying all values by 10 it gives then same, and pushing all values lower than 1sec to 1sec, it still fails with the same timeout error.
In the wireshark dump, it seems the transmission ends on a command from the master (`12:10:10:10:10:10`), and the slave (`10:10:10:10:10:10`) is not responding, so maybe one of the packets previously sent by the master is malformed ?

jamwaffles, 2023-05-13:
FWIW, your MAC address comment is backwards; the master sends frames with `10:10:...` specified with the `ETHERCAT_ETHERTYPE` const. The first slave device in the chain will set the [U/L bit](https://en.wikipedia.org/wiki/MAC_address#Universal_vs._local_(U/L_bit)) resulting in `12:10:...` in the returned frames.

Thanks for the captures and log output. I can see that the slave device never sets the "mailbox written" flag when timing out, i.e. [this check](https://github.com/ethercrab-rs/ethercrab/blob/master/src/slave/mod.rs#L255). Nothing obvious stands out to me so far as to _why_ it wouldn't set that though :(

Have you done anything to rule out a hardware problem? I'm by no means saying EtherCrab is perfect and the bug isn't in my code but it's worth reordering the drives and/or cables to rule out a broken drive/port/wire.

Also, if you have it set up, would you be able to run the `slaveinfo` example from SOEM and post the output and a wireshark capture here? It would be good to compare against a known working library.

If all else fails, I can scrape together 4 drives in a chain and see if I can reproduce the problem, although that would be 3x EC400 and 1x AKD so not the same hardware as you.

jimy-byerley, 2023-05-15:
> FWIW, your MAC address comment is backwards

Hum .. you are right

> Have you done anything to rule out a hardware problem?

I tried to swap the omron servodrives, but I cannot easily swap the omron servodrives with the zeroerr servodrives because of the machine design.
swaping the omron drives did not helped.  My omron PLC however is successfully communicating with all them all. I don't know what is the ethercat implementation of the PLC though ...

jimy-byerley, 2023-05-15:
I will give the `slaveinfo` example a try

jimy-byerley, 2023-05-15:
Here is the result of `slaveinfo`: [dump-timeout.zip](https://github.com/ethercrab-rs/ethercrab/files/11480249/dump-timeout.zip)

fpdotmonkey, 2023-06-28:
I'm having a similar issue where there's a crash during initialization, possibly while the FMMU is being configured.  My slave setup is a EK1100-EL4132-EL3104, all Beckhoff hardware.  I also applied a small patch to make the EK1100 demo work on my Windows machine, as well some special stuff for the EL3104.  The patch, trace logs during execution, and a wireshark capture of that execution, are in the below ZIP.
[ethercrab-init-crash.zip](https://github.com/ethercrab-rs/ethercrab/files/11895418/ethercrab-init-crash.zip)

jamwaffles, 2023-06-29:
Note to self: I can reproduce the timeout error message (but who knows if it's the same issue) with two Leadshine EL7 drives in a chain. A single drive doesn't error.

PR #57 (2023-07-04): Fix slave init timeout 
I couldn't find anything in the specs for a root cause, but it seemed to timeout reading `0x1c00` either repeatedly, or once the slave has been in PRE-OP for a while or... something. I'm not entirely sure which is scary.

It seems to work with the changes in this PR - I can now get two EC400 servo drives to get past the timeout.

@jimy-byerley I appreciate you might've moved on from EtherCrab entirely, but if you have some time would you mind testing this PR against your setup? I'd like to ask the same of @fpdotmonkey too if you can.

E.g. in `Cargo.toml`:

```toml
ethe [... cut]