ethercrab-rs/ethercrab #53: Storage initialization possibly not zeroing backing store properly

agalauner-r7, 2023-06-03:
Hi,

I am currently reading a bit of your code because I am curious how it works and I crossed the following line in [src/pdu_loop/storage.rs](https://github.com/ethercrab-rs/ethercrab/blob/master/src/pdu_loop/storage.rs#L60-61):

    // MSRV: Use MaybeUninit::zeroed when `const_maybe_uninit_zeroed` is stabilised.
    let frames = UnsafeCell::new(MaybeUninit::uninit());

This ideally wants to initialize the `frames` slice wrapped in the `UnsafeCell` and `MaybeUninit`, but because `const_maybe_uninit_zeroed` isn't stable yet, you treat it as completely uninitialized NOT ZEROED memory. So far, so good.

A bit further down in [line 93](https://github.com/ethercrab-rs/ethercrab/blob/master/src/pdu_loop/storage.rs#L93), you are attempting to zero the memory of the slice by getting a mutable pointer out of the `UnsafeCell` and then out of the `MaybeUninit` and then you are writing ONE zero byte into that location using `write_bytes(0u8, 1)`:

    // MSRV: Remove when `const_maybe_uninit_zeroed` is stabilised. Rely on
    // `MaybeUninit::zeroed` in `PduStorage::new()`.
    unsafe { (&mut *self.frames.get()).as_mut_ptr().write_bytes(0u8, 1) };

Shouldn't that call rather write `core::mem::size_of::<[PacketBuffer<DATA>; N]>()` zero bytes to entirely zero the backing store? As far as I understand it, `MaybeUninit::zeroed()` should fill the whole slice with all its elements to zero bytes.

I might be wrong there, but I think these two approaches do different things. I also don't understand what the benefit of just zeroing the first byte of the slice has apart from setting the first byte of a `FrameElement` in the slice which should be in turn the first byte of a `PduFrame` which is the `index` as a u8.

Also, why do you do all this in `as_ref()` and not in `new()` when you would call `MaybeUninit::zeroed()` there anyway later? Why not place that pointer hack in `new` right now until `const_maybe_uninit_zeroed` is stable?

jamwaffles, 2023-06-03:
Hey, thanks for the question! `write_bytes` has tripped me up a couple of times too, but I believe the behaviour in the code is correct. As [the documentation](https://doc.rust-lang.org/stable/std/primitive.pointer.html#method.write_bytes) states:

> Invokes memset on the specified pointer, setting `count * size_of::<T>()` bytes of memory starting at `self` to `val`.

`T` in this case is a `[FrameElement<DATA>; N]` so the `1` is the correct value here and will zero the entire array.

> Also, why do you do all this in as_ref() and not in new()

I'd like to do it in `new()`, but a lot of these methods aren't stably `const` yet, and I'd like `PduLoop::new` to remain `const` so we can make `'static` instances of `PduLoop` easily. `as_ref` is called in non-static and non-const contexts so must remain seperate, at least for now.

agalauner-r7, 2023-06-03:
Oh wow! I totally missed that bit in the documentation. I feel a bit dumb now...

And I also missed that `new()` is actually `const` and that you use `PduStorage` in a static variable in all the examples. Now that you said it, that also makes perfect sense.

Thank you for the explanation!

closed 2023-06-03 by agalauner-r7

jamwaffles, 2023-06-03:
No problem! Glad I could clear things up a bit. It was a good double check for me as well, so definitely a benefit to asking this question :)