ethercrab-rs/ethercrab #56: smoltcp feature phy-raw_socket requires std

theo-lightworks, 2023-06-28:
https://github.com/ethercrab-rs/ethercrab/blob/a1edeb8da18d19fffeebdd3cca8201e1b79d99a6/Cargo.toml#L36C126-L36C126

see 

https://github.com/smoltcp-rs/smoltcp/blob/a6841955707112d68bb87767e2397af4d5a6c34b/Cargo.toml#L45C16-L45C16

jamwaffles, 2023-06-28:
Hey, thanks for opening the issue. Can you describe what the problem is a bit more please? I'm not sure I understand. Are you saying that `ethercrab` should only enable `smoltcp/phy-raw_socket` when using _its_ `std` feature, or something else?

theo-lightworks, 2023-07-03:
Hi, I was interested in testing ethercrab in a no_std environment, but one of the issues with that is that it is not possible to compile ethercrab for no_std as the phy-raw_socket feature of smoltcp is set as a required dependency of ethercrab.

jamwaffles, 2023-07-03:
Ah I think I see now, thanks! I've opened a fix in #58 which should fix this issue.

jamwaffles, 2023-07-03:
Hey @theo-lightworks if you'd like to test the most recent master commit, it should hopefully fix your `no_std` issue. It's also now tested in CI so hopefully no regressions either!

closed 2023-07-03 by jamwaffles

theo-lightworks, 2023-07-03:
Hi @jamwaffles It does fix the build, but I have some issues related to async. I tried it with edge-executor, not sure if I am just doing something incorrectly though.

```
error: future cannot be sent between threads safely
   --> src/lib.rs:30:13
    |
30  |       e.spawn(async {
    |  _____________^
31  | |         setup().await;
32  | |     });
    | |_____^ future created by async block is not `Send`
    |
    = help: within `ethercrab::slave_group::configurator::SlaveGroupRef<'_>`, the trait `Sync` is not implemented for `UnsafeCell<ethercrab::slave_group::configurator::GroupInnerRef<'_>>`
note: future is not `Send` as this value is used across an await
   --> /home/theo/source/ethercrab/src/client.rs:263:73
    |
263 |             offset = unsafe { group.configure_from_eeprom(offset, self).await? };
    |                               -----                                     ^^^^^   - `group` is later dropped here
    |                               |                                         |
    |                               |                                         await occurs here, with `group` maybe used later
    |                               has type `&ethercrab::slave_group::configurator::SlaveGroupRef<'_>` which is not `Send`
note: required by a bound in `Executor::<'a, C, M, S>::spawn`
```

jamwaffles, 2023-07-03:
Glad to hear it fixes the build at least :). Do you have a full code example or a repository I could look at somewhere? It's difficult to see what the error might be without more context.

flaminggoat, 2023-07-03:
I made an example here, I've tried to use libraries that would work on no_std. It works with one executor but not the other https://github.com/flaminggoat/ethercat_no_std

PR #58 (2023-07-03): Fix smoltcp feature
Also fixes usage of `Box` in no_std so this crate should now compile for no_std targets. Note though that it still requires `alloc` to be present.

I'm hoping to redesign the API a bit to remove the requirement on `Box<dyn ...>` which would mean no dynamic alloc is required by EtherCrab at all.

Closes #56