ethercrab-rs/ethercrab #98: Redundant network interface support

jamwaffles, 2023-08-14:
EtherCAT supports redundant interfaces where a second NIC is attached to the other end of the network. EtherCrab should support this feature.

keenanjohnson, 2024-04-23:
Cable Redundancy
Cable redundancy is designed to compensate for the failure of a communication cable section in the EtherCAT system. A ring topology, which normally is operated in both directions, is therefore used. Both branches can nevertheless still be reached if the ring is interrupted at some point.

[image]

Description
A second network port is used for ring closure at the EtherCAT master control system. Both cyclic and acyclic frames are sent simultaneously through both ports, and are transported through the system.

In the absence of any fault, all the EtherCAT slaves are reached in the forward direction (so called processing direction) from the primary port. This means that they are processed, since the EtherCAT Slave Controller (ESC) is only passed through in the forward sense.
When there is no fault, all the EtherCAT slaves are reached from the secondary port in the reverse direction - the data in the "redundancy" frame is therefore not changed.

In each case, the EtherCAT frames arrive, possibly modified, at the other port, and are checked by the EtherCAT master. In case of a cable break, both frames are processed - each one on the respective side of the failure. Therefore both frames contain a part of the input data. The master has to combine the data of both frames, and gets one frame with all the input data. The working counters from both frames are added to check for its validity. It is unimportant whether an EtherCAT slave is reached from the primary or redundancy port. The EtherCAT master has to consider that a frame on one side is lost and the other frame returns. To find the matching frame it is useful to mark the frames with an identification or use appropriate mechanism.

The cable redundancy is single-error tolerant, i.e. communication with the slaves can continue if the cable is interrupted in one place. When the communication is restored the original communication direction is restored. If the communication is interrupted in more than one place, all connections have to be restored before another fault may occur.

Functionality
In case of cable break all types of EtherCAT communications (process data and mailbox protocols) are be supported without any restrictions.
Handling of the following use cases:

Normal operation
Stay operational in case of cable break between two slaves
Stay operational in case of cable break between primary port and first slave
Stay operational in case of cable break between secondary port and last slave
Stay operational in case of cable fixed
Start/Stop (State change) in case of cable break
Adjustment of Auto Increment address in case of cable break
Frame loss in case of cable break (partner frame was not received)

keenanjohnson, 2024-04-24:
Are there useful bite sized chunks that community contributors like me could start biting off to make this happen?

jamwaffles, 2024-04-24:
Off the top of my head, not really I'm afraid. I haven't thought about what's involved for EtherCrab to support this, but I can imagine some decently sized refactors being required that can't be broken down that easily.

jamwaffles, 2025-02-14:
Note to self: good explanation of redundancy [here](https://github.com/OpenEtherCATsociety/SOEM/issues/621#issuecomment-1167837786).

jbbjarnason, 2025-03-04:
For future reference, I was integrating redundancy to our C++ ethercat executable with soem and I heavily missed some indication where the circle is broken, I could get indication that the circle is broken but no way on determining where. 
If or when this feature is implemented it would be awesome to include this, I don't know the specifics of how this would be integrated but soem as a concept/terminology of stacknumber (0=primary 1=secondary stack) which could be included in per device frame.

jamwaffles, 2025-03-04:
> If or when this feature is implemented it would be awesome to include this

Yeah that would be great! I can imagine it would make fault finding much easier. I'll bear this in mind when I work on this feature.

Reisande, 2025-05-15:
@jbbjarnason I have experience with cable redundancy in ethercat. Is there any way I would be able to contribute to this feature/debug the lost the broken location issue?

jbbjarnason, 2025-05-15:
I was referring to a flaw in the [soem](https://github.com/OpenEtherCATsociety/SOEM) library, ethercat redundancy is still a feature missing from ethercrab i believe. So there is no problem with ethercrab, just a missing feature.

Reisande, 2025-05-15:
Ah ok, @jamwaffles  is there any work I can do to help implement this feature?