ethercrab-rs/ethercrab #215: Problem understanding IOs

kauanfontanela, 2024-06-14:
I can't quite understand how TxPdos are ordered in `SlaveRef.inputs_raw`. Take the EL1809 ESI file description for input channels 1 and 2
```xml
<TxPdo Fixed="1" Mandatory="1" Sm="0">
	<Index>#x1a00</Index>
	<Name>Channel 1</Name>
	<Entry>
		<Index>#x6000</Index>
		<SubIndex>1</SubIndex>
		<BitLen>1</BitLen>
		<Name>Input</Name>
		<DataType>BOOL</DataType>
	</Entry>
</TxPdo>
<TxPdo Fixed="1" Mandatory="1" Sm="0">
	<Index>#x1a01</Index>
	<Name>Channel 2</Name>
	<Entry>
		<Index>#x6010</Index>
		<SubIndex>1</SubIndex>
		<BitLen>1</BitLen>
		<Name>Input</Name>
		<DataType>BOOL</DataType>
	</Entry>
</TxPdo>
```

Their entries are in addresses `0x6000` and `0x6010`. How come they are one bit apart in the `SlaveRef.inputs_raw` bytes?
```rs
SlaveRef.inputs_raw[0] // 0b00000001 -> Channel 1
SlaveRef.inputs_raw[0] // 0b00000010 -> Channel 2
```

What am i missing? Does ethercrab provide some way to acces entries via their index and subindexes?

jamwaffles, 2024-06-15:
The `<BitLen>1</BitLen>` means the values are (or can be) packed into bytes, so that's how the EL1809 will present the input states in the PDI. 

> Does ethercrab provide some way to acces entries via their index and subindexes?

It doesn't, but the PDI doesn't correspond to the PDO mappings, so this doesn't make much sense. That said, maybe I'm missing a use case you're looking for - can you explain a bit more why you'd like to access the EL1809 inputs via their PDO/SDO?

kauanfontanela, 2024-06-17:
Thank you for responding.
I'm looking for developing a master program that can run with any input module as a slave. My goal is to make it easy for the final user to assign the roles of each input channel (like in TwinCAT, for instance), and for that I rely on the ESI files of each device so that I can correctly reference their addresses. However, as you said, it seems like the PDI doesn't correspond to PDO mappings. Now I'm struggling to understand why are there PDO mappings, and how other projects like [EtherCAT.NET](https://github.com/Apollo3zehn/EtherCAT.NET/) can get the values from PDI referencing by their PDO address.

jamwaffles, 2024-06-18:
Thanks for the explanation :)

PDOs are used to select which data is to be mapped into the PDI via every SubDevice's FMMU(s). The PDI is a single contiguous address space for efficiency reasons. Each SubDevice reads/writes into this space at a configured address to send/receive data. At this point, the PDOs and their meaning are lost - it's just bytes - it's up to the MainDevice and SubDevice to recreate the semantic meaning behind the bytes on each end. 

For more fixed applications, this is achieved in code on the MainDevice using e.g. a `struct` and [`ethercrab-write-derive`](https://docs.rs/ethercrab-wire-derive) as one option.

To build a dynamic UI like TwinCAT, you'll need to use the ESI to both set the PDOs during startup, as well as re-map the PDI bytes back to their semantic fields (PDO addresses, names, etc) at runtime.

kauanfontanela, 2024-07-16:
> To build a dynamic UI like TwinCAT, you'll need to use the ESI to both set the PDOs during startup, as well as re-map the PDI bytes back to their semantic fields (PDO addresses, names, etc) at runtime.

Sorry, I am still struggling to find out how I would re-map the PDI bytes back to their PDO address. 
Currently I'm getting the TxPdo Index and subtracting `0x1a00`, than wrapping it around 8 to get the index and bit of the `inputs_raw` array, but this feels way to hacky and inconsistent. 
```rs
let pdo_idx = pdo_idx - 0x1a00;
let idx: usize = (pdo_idx / 8).into();
let bit = pdo_idx % 8;

let input = inputs_raw[idx] & (1 << bit);
let boolean = input != 0
```

jamwaffles, 2024-07-16:
You need to keep the mapping in your application, hence my suggestion to use the ESI file to help with this. There's no direct way to map the PDI back to what was configured with `sdo_write` etc during init - you'll need to keep a record of which fields are configured in your application, and use that to map back to the semantic fields in the UI.

Subtracting `0x1a00` from the index probably just _happens_ to be working, e.g. you might need to configure PDOs in `0x1a01` as well, or have multi-byte fields, which would break this assumption.

The best supported way to do this right now is statically, with a `struct` and [`ethercrab-wire-derive`](https://docs.rs/ethercrab-wire-derive/latest/ethercrab_wire_derive/). I'll give an example using it below. Although this won't work for your application if you want to support any SubDevice dynamically, it will hopefully give some insight into how the EtherCAT side of this works.

First, the struct is defined as such:

```rust
#[derive(ethercrab::EtherCrabWireWrite, Debug)]
#[wire(bytes = 11)]
pub struct SubDeviceData {
    #[wire(bytes = 2)]
    pub status: u16,
    #[wire(bytes = 1)]
    pub op_mode: u8,
    #[wire(bytes = 4)]
    pub actual_position: i32,
    #[wire(bytes = 4)]
    pub actual_velocity: i32,
}
```

The `EtherCrabWireWrite` macro will generate some code that correctly calculates the offsets into a slice to decode each field from. In pseudo-Rust, it kind of looks like this:

```rust
fn unpack_drive_outputs(pdi: &[u8]) -> SubDeviceData {
    let status = u16::from_le_bytes(&pdi[0..2]);
    let op_mode = u8::from_le_bytes(&pdi[2..3]);
    let actual_position = i32::from_le_bytes(&pdi[3..7]);
    let actual_velocity = i32::from_le_bytes(&pdi[7..11]);

    SubDeviceData {
        status,
        op_mode,
        actual_position,
        actual_velocity,
    }
}
```

The PDOs then need to be configured properly during init:

```rust
// Zero out object count ready for new config
sd.sdo_write(0x1a00, 0, 0u8).await?;
// Status word, u16
sd.sdo_write(0x1a00, 1, 0x6041_0010u32).await?;
// Operation mode display, u8
sd.sdo_write(0x1a00, 2, 0x6061_0008u32).await?;
// Actual position, i32
sd.sdo_write(0x1a00, 3, 0x6064_0020u32).await?;
// Actual velocity, i32
sd.sdo_write(0x1a00, 4, 0x606c_0020u32).await?;
// 4 fields assigned to this PDO mapping
sd.sdo_write(0x1a00, 0, 0x04u8).await?;
```

Hopefully it's clear now that there's no direct way to "un-map" the PDI back into semantic fields - it has to be done by the programmer (or in the above case, the derive macro) to make sure the `struct` has the same representation as the PDOs we map using the several `sdo_write` calls.

This is where the ESI file comes in. You could use that to pass values to `sdo_write` to configure which PDOs are in which order/offset in the PDI. You could _also_ then use it to create some form of mapping the PDI `&[u8]` slice back into a `struct`, `HashMap`, or whatever representation you want. Remember the PDI is _just_ bytes - it's up to how you configure the PDO mappings, and how you decode those bytes back into a more semantic meaning.

Hopefully this makes things a little clearer. I'd very much like to be able to generate a lot of this code using a `derive` and an ESI file in the future, but that's a bit of a nice-to-have feature right now and I don't have much time to implement it.

kauanfontanela, 2024-07-17:
Thank you for making it clear!

Although, I still cant manage to remap the PDOs because of a `NoMailbox` error;

```rs
let group = client
    .init_single_group::<MAX_SLAVES, PDI_LEN>(ethercrab::std::ethercat_now)
    .await
    .expect("Init");

for slave in group.iter(&client) {
    if slave.name() == "EL1809" {
        let status = slave.status().await?;
        println!("slave {} {} {}", slave.name(), status.0, status.1);
        slave.sdo_write(0x1a00, 0, 0u8).await?;
    }
}
```

result:
```
slave EL1809 Pre-Operational 0x0000: No error
... ERROR ethercrab::slave] No write (slave IN) mailbox found but one is required
```

I tried `sdo_write` on all states without success. I also found this startup table in the [EL18xx documentation](https://download.beckhoff.com/download/Document/io/ethercat-terminals/EL18xxen.pdf) and blindly tried the indexes from this table, but still got the same error. I cant read any SDO either.
[image]

jamwaffles, 2024-07-17:
Ah great to hear it's a bit clearer now! As for the SDO read/writes failing: the EL1809 is a simple SubDevice so, while it does have SDOs, the PDOs can't be configured. The ESI file has `Fixed="1"` which I think is what defines that, but also [the documentation](https://infosys.beckhoff.com/english.php?content=../content/1033/el18xx/1670568203.html&id=5668441045142890787) states that:

[image]

You might be able to read `0x1c12` and `0x1c13` like TwinCAT, but my guess is TC is getting that data from the ESI file. What happens if you `sdo_read(0x1c12)`?

kauanfontanela, 2024-07-17:
Reading `0x1c12` gives the same **write** error.
```rs
if slave.name() == "EL1809" {
    let status = slave.status().await?;
    println!("slave {} {} {}", slave.name(), status.0, status.1);

    let read = slave.sdo_read::<u16>(0x1c12, 0).await?;
    println!("0x1c12: {:?}", read);
}
```
```
slave EL1809 Pre-Operational 0x0000: No error
... ERROR ethercrab::slave] No write (slave IN) mailbox found but one is required
```

How could TwinCAT map the PDI value to the PDO addresses without mailbox? Could it be some kind of convention among Beckhoff devices?

kauanfontanela, 2024-07-17:
Guess I will try a different approach. Thank you for the attention!

Lomsor, 2024-07-17:
This is how an EL1809 looks like in my TwinCAT 3.1.2026.6:
[image]
[image]
For devices without CoE the relevant tabs aren't shown.

Like mentioned by @jamwaffles, the PDOs for devices like this are defined as fixed in the ESI file, seen as referenced in the screenshot above. TwinCAT uses this information in the config stage to define what data to expect from communication. It then uses this to build a representation of the expected structure and how it would map it to internal memory on the device running the runtime. This allows to map/link to variables defined in modules, like an SPS module.
But what that really does is the runtime copying from one defined runtime memory addresses to another as configured. The EtherCAT is processed according to what is configured in the "SyncUnits". The default behaviour is that the EtherCAT sync for specific I/Os is run after a linked PLC Task is done processing, but this can be customized to some extent in TwinCAT.

It does the same with configurable I/O Modules, with the difference that the process data is changed by configuring the I/O hardware module itself during startup. The variable linking is still done to specific memory sections calculated by TwinCAT.

The ">Address" shown in TwinCAT is an address on the TwinCAT Runtime memory space, not a position on an EtherCAT frame. The frame itself can't be seen in TwinCAT itself, as far as I know. Only the size of the frame, the hardware order and the final image on the runtime memory.

jamwaffles, 2024-07-17:
Thanks very much for the TwinCAT description @Lomsor. 

@kauanfontanela hopefully this issue is closed out of understanding and not frustration :)

kauanfontanela, 2024-07-17:
> But what that really does is the runtime copying from one defined runtime memory addresses to another as configured.

I cant see how the ESI info is helpful in finding those addresses since it is not compatible with the PDI indexes. Is this behavior specific for Beckhoff devices for there is some assumption about indexes?

***ps***: sorry, I closed the issue exactly when @Lomsor commented :sweat_smile: . Thank you @jamwaffles, I had closed the issue for I figured I would try another approach for my problem

Lomsor, 2024-07-17:
> > But what that really does is the runtime copying from one defined runtime memory addresses to another as configured.
> 
> I cant see how the ESI info is helpful in finding those addresses since it is not compatible with the PDI indexes.

If you mean the adresses I meant in the quote of mine, the TwinCAT Runtime memory adresses,  those won't be helpfull here as those are TwinCAT specific and don't apply.
TwinCAT is supposed to work with other EtherCAT hardware as well, although personally I haven't tried myself.

As for how to map the EtherCAT frame to the PDOs in the ESI file, I think we are missing some documentation regarding what each entry in the ESI means. I'm not a registered member in the EtherCAT Tech Group and the Beckhoff documentation doesn't go into detail here but I found this poster that has some info in the top right:
https://www.ethercat.org/download/documents/EtherCAT_Device_Protocol_Poster.pdf

I've also stumbled upon a rust crate that handles ESI files.

Without (being able) to read the documentation it all points towards the ESI file being mostly for device internal mappings. With both of the indexes in the file corresponding to device internal adresses.
I guess the EtherCAT Slave Controller (ESC) just reads/writes sequentially from/to the EtherCAT frame acording to the configuration from/to the device memory. This would align with how EtherCAT works. The gaps in the Indexes in Beckhoff ESI indicate that there is unused memory between data on the slave devices, which probably is a concious choice made by the vendors.

It seems like only the PDO ordering, datatype and size as well as defined spacing is considered in the master.

Of course, EtherCrabs handlig of the data structure is to be considered.

jamwaffles, 2024-08-13:
I'm doing a bit of issue cleanup and as this thread hasn't had a reply in a month I'm going to close it. If there's still something to be said, do please re-open the issue to continue the discussion!

closed 2024-08-13 by jamwaffles