ethercrab-rs/ethercrab #249: Control word disable voltage

jbbjarnason, 2024-11-19:
fyi, I am pretty certain that DISABLE_VOLTAGE should be ENABLE_VOLTAGE

from ds402.rs
```rust
        /// Switch on
        const SWITCH_ON = 1 << 0;
        /// Disable Voltage
        const ENABLE_VOLTAGE = 1 << 1; // <----------------------------
        /// Quick Stop
        const QUICK_STOP = 1 << 2;
        /// Enable Operation
        const ENABLE_OP = 1 << 3;
        /// Operation mode specific
        const OP_SPECIFIC_1 = 1 << 4;
        /// Operation mode specific
        const OP_SPECIFIC_2 = 1 << 5;
        /// Operation mode specific
        const OP_SPECIFIC_3 = 1 << 6;
        /// Reset Fault (only effective for faults)
        const RESET_FAULT = 1 << 7;
        /// Pause/halt
        const PAUSE = 1 << 8;

        /// Shutdown state.
        const STATE_SHUTDOWN = Self::QUICK_STOP.bits() | Self::DISABLE_VOLTAGE.bits();
        /// Switched on state.
        const STATE_SWITCH_ON = Self::QUICK_STOP.bits() | Self::DISABLE_VOLTAGE.bits() | Self::SWITCH_ON.bits();
        /// Voltage disabled state.
        const STATE_DISABLE_VOLTAGE = 0;
        /// Quick stop state.
        const STATE_QUICK_STOP = Self::DISABLE_VOLTAGE.bits();
        /// Operation disabled state.
        const STATE_DISABLE_OP = Self::QUICK_STOP.bits() | Self::DISABLE_VOLTAGE.bits() | Self::SWITCH_ON.bits();
        /// Operation enabled state.
        const STATE_ENABLE_OP = Self::ENABLE_OP.bits() | Self::QUICK_STOP.bits() | Self::DISABLE_VOLTAGE.bits() | Self::SWITCH_ON.bits();
        /// Fault reset state.
        const STATE_FAULT_RESET = Self::RESET_FAULT.bits();
```

jamwaffles, 2024-11-19:
You're very probably correct, although I wouldn't spend much time fixing the builtin DS402 implementation - it hasn't seen much love recently and was only ever a prototype. I think I made a mistake in releasing it with EtherCrab.

That said, I'll soon be focussing on a better reimplementation that's better built and better thought out, so stay tuned (ish - unsure what the timeline is) for that.

jamwaffles, 2025-02-07:
Just doing a bit of housekeeping. I'll close this issue in favour of #233.