ethercrab-rs/ethercrab #250: Unpack from slice

jbbjarnason, 2024-11-20:
I changed type of bar to u32 and get `ReadBufferTooShort`

```bash
thread 'basic_struct' panicked at ethercrab-wire/tests/unpack.rs:28:46:
called `Result::unwrap()` on an `Err` value: ReadBufferTooShort
stack backtrace:
   0: rust_begin_unwind
             at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/std/src/panicking.rs:652:5
   1: core::panicking::panic_fmt
             at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/panicking.rs:72:14
   2: core::result::unwrap_failed
             at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/result.rs:1654:5
   3: core::result::Result<T,E>::unwrap
             at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/result.rs:1077:23
   4: unpack::basic_struct
             at ./tests/unpack.rs:28:15
   5: unpack::basic_struct::{{closure}}
             at ./tests/unpack.rs:4:18
   6: core::ops::function::FnOnce::call_once
             at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ops/function.rs:250:5
   7: core::ops::function::FnOnce::call_once
             at /rustc/129f3b9964af4d4a709d1383930ade12dfe7c081/library/core/src/ops/function.rs:250:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

failures:
    basic_struct
```

```diff
diff --git a/ethercrab-wire/tests/unpack.rs b/ethercrab-wire/tests/unpack.rs
index 408ad53..0b5490a 100644
--- a/ethercrab-wire/tests/unpack.rs
+++ b/ethercrab-wire/tests/unpack.rs
@@ -8,7 +8,7 @@ fn basic_struct() {
         #[wire(bits = 8)]
         foo: u8,
         #[wire(bits = 16)]
-        bar: u16,
+        bar: u32,
         #[wire(bits = 32)]
         baz: u32,
     }
```

I am guessing this should work?

jamwaffles, 2024-11-20:
Hm, yes that should still work because `#[wire(bits = 16)]` is forcing the length. Thanks for the bug report! I'll take a look when I have a mo :)

jamwaffles, 2024-11-28:
Thinking about this a bit more, I'm hesitant to call this a bug. Perhaps it would be useful to have another derive attribute instead. Something like e.g.

```rust
struct Example {
    /// Decode the lower 2 bytes of the field, padding the rest with zeros
    #[wire(bits = 16, pad = 16)]
    field: u32

    /// Equivalent in bytes
    #[wire(bytes = 2, pad_bytes = 2)]
    equivalent: u32
}
```

I'm still wondering why you need to decode a `u16`'s worth of bytes into a `u32` though. Are there two bytes after the field you'd like to discard or something? Would `#[wire(post_skip_bytes = N)]` help here?

jbbjarnason, 2024-11-29:
I just stumbled upon this when I was trying to unpack a fixed size array of u8, but simplified the example here.
Afterwards I could not even use array of u8, since my data from the device was packed on came on bit level and I didn't know how to declare how to transpose bit level to array of u8 or bool.

So I made a workaround using bitview in process loop. 

I agree this is a feature.

jamwaffles, 2025-03-28:
I'm going to close this as "not intending to implement". I think the best way to extract a bitfield is to use a nested `struct` that holds all of the bitfields, or you could use a `[u8; 2]` if you don't mind a bit of bit twiddling or need the ergonomics of the struct. Thanks for the discussion!

closed 2025-03-28 by jamwaffles