ethercrab-rs/ethercrab #255: Possible soundness issue with PDI buffer accesses

Rahix, 2024-12-12:
Hi,

I am currently trying to understand how ethercrab manages accesses to the PDI buffer between the user application and the bus driver.  I think I may have stumbled over a soundness violation, which I am looking to clarify.

Some background: In the ethercrab examples, the application cycle is always synchronized with the bus cycle from what I understand.  Calling `rx_tx()` on the group pauses the application cycle until the bus update has occurred.  For a lot of usecases this makes sense, of course.

However, I was trying to understand whether ethercrab also supports keeping application and bus cycle asynchronous.  So the application continuosly updates output values in the PDI and the bus picks up whatever values are available at the time it gets around to updating the I/O again.

In my naive understanding, I thought this could be possible by just moving the group's `rx_tx()` call into a separate async task.  And giving this a dry attempt, it seems to compile:

```rust
// Modified examples/ek1100.rs

let maindevice = std::sync::Arc::new(maindevice);
let group = std::sync::Arc::new(group);

tokio::spawn({
    let maindevice = maindevice.clone();
    let group = group.clone();
    async move {
        loop {
            group.tx_rx(&maindevice).await.expect("TX/RX");
        }
    }
});

loop {
    {
        let mut subdevice = group.subdevice(&maindevice, 0).unwrap();
        let (_i, o) = subdevice.io_raw_mut();
        for byte in o.iter_mut() {
            *byte = byte.wrapping_add(1);
        }
    }

    tick_interval.tick().await;
}
```

I do not have any Ethercat hardware so I was not able to test whether this works at all or run the sample with MIRI to see whether it triggers.  But from reading the code, I think this leads to an aliasing violation.

Both the `tx_rx()` and the `subdevice()` create slices of the PDI outputs, with the one created by `subdevice()` being mutable.  By having both present at the same time, aliasing rules are violated.

It also seems problematic to me from a logic standpoint:  As the `tx_rx()` could copy data from the PDI at any time, there is no guarantee on the consistency of the values transmitted.

If this all is true, I think the correct way to implement my usecase would be a variant of `tx_rx()` that is still called in-sync with the application cycle but does not wait for an actual bus update to have occurred.  Additionally, the existing `tx_rx()` would need an additional safeguard to ensure no `subdevice()` borrows of the PDI are still alive (e.g. by making `tx_rx()` take `&mut self`?)

Am I completely off track?  Or is there really a problem here?  Would love to get more insight :)

## Environment

- Version of ethercrab in use or git hash: 8f07c522a458 ("Add blocking TX/RX task advice")
- Operating system kind and version: Linux (not applicable)
- EtherCAT devices in use: none, only compile-tested

jamwaffles, 2024-12-12:
Thanks for taking the time to think about and report this! If that code compiles, it shouldn't! I thought I'd got the API pretty watertight in terms of this, but it appears not.

I've only skimmed the issue for now but I'll dig into this more later when I've got a bit of time. I think you could be right in that I just need to sprinkle a few `&mut Self`s here and there.

FWIW my intent for the API was that if the user wants to read/write the PDI in another thread/task, they need to bring their own synchronisation primitives, so everything provided by EtherCrab should be `Send + !Sync` (if I've got those bounds right, i.e. usable in one thread only).

Rahix, 2024-12-12:
> I think you could be right in that I just need to sprinkle a few &mut Selfs here and there.

Unfortunately, it doesn't seem to be that simple.  I tried making `tx_rx()` take `&mut self` to model the mutable access to the whole PDI.  Some examples can be modified to get this to work:

<details>
<summary>Example change to <code>multiple-groups.rs</code> (click to expand)</summary>

```diff
--- a/examples/multiple-groups.rs
+++ b/examples/multiple-groups.rs
@@ -92,7 +92,7 @@ async fn main() -> Result<(), Error> {
     let maindevice_slow = maindevice.clone();
 
     let slow_task = tokio::spawn(async move {
-        let slow_outputs = slow_outputs
+        let mut slow_outputs = slow_outputs
             .into_op(&maindevice_slow)
             .await
             .expect("PRE-OP -> OP");
@@ -105,14 +105,16 @@ async fn main() -> Result<(), Error> {
         // Only update "slow" outputs every 250ms using this instant
         let mut tick = Instant::now();
 
-        // EK1100 is first SubDevice, EL2889 is second
-        let mut el2889 = slow_outputs
-            .subdevice(&maindevice_slow, 1)
-            .expect("EL2889 not present!");
-
-        // Set initial output state
-        el2889.io_raw_mut().1[0] = 0x01;
-        el2889.io_raw_mut().1[1] = 0x80;
+        {
+            // EK1100 is first SubDevice, EL2889 is second
+            let mut el2889 = slow_outputs
+                .subdevice(&maindevice_slow, 1)
+                .expect("EL2889 not present!");
+
+            // Set initial output state
+            el2889.io_raw_mut().1[0] = 0x01;
+            el2889.io_raw_mut().1[1] = 0x80;
+        }
 
         loop {
             slow_outputs.tx_rx(&maindevice_slow).await.expect("TX/RX");
@@ -121,6 +123,10 @@ async fn main() -> Result<(), Error> {
             if tick.elapsed() > slow_duration {
                 tick = Instant::now();
 
+                let mut el2889 = slow_outputs
+                    .subdevice(&maindevice_slow, 1)
+                    .expect("EL2889 not present!");
+
                 let (_i, o) = el2889.io_raw_mut();
 
                 // Make a nice pattern on EL2889 LEDs
```

</details>

However, other uses cannot be adapted.  For example, the `ec400.rs` example takes a long-lasting borrow of the PDI very early and then later uses both that borrow and the `tx_rx()` call interleaved:

```rust
let subdevice = group.subdevice(&maindevice, 0).expect("No servo!");
let mut servo = Ds402Sm::new(Ds402::new(subdevice).expect("Failed to gather DS402"));

// ...

loop {
    group.tx_rx(&maindevice).await.expect("TX/RX");

    // ...

    let (i, o) = servo.subdevice().io_raw_mut();
}
```

Technically, this is already unsound use, I think?  Because to my understanding the subdevice holds a mutable slice while `tx_rx()` is called?

jamwaffles, 2024-12-12:
> Technically, this is already unsound use

Indeed, although mercifully the accesses don't overlap but that's purely due to the ordering of the logic. 

Thanks for posting another example - they will be useful for writing miri/compilefail tests.

jamwaffles, 2024-12-15:
This is partially fixed in #258, in the sense that it should no longer be possible to hold a reference to a SubDevice across a `tx_rx*` call. I believe this removes that unsoundness hole, but as you saw breaks the `Ds402` architecture, so I've just removed that completely (for now). From the start, groups should only have been accessible in a single task/thread unless you bring your own synchronisation primitives. This should actually be the case one #258 is merged.

PR #258 (2024-12-15): Make all `tx_rx*` methods `&mut Self`
`SubDeviceGroup`, whilst implementing `Sync` for compatibility with some executor runtimes, is unsound as it currently stands. This is because a `SubDeviceRef` can access its PDI concurrently with `tx_rx*` without any kind of synchronisation. Making `tx_rx*` `mut` prevents these references being held at the same time.

Groups should be isolated to a single task/thread. You need to bring your own synchronisation primitives to share it across threads (`RwLock`, etc).

This PR breaks the DS402 stuff, so I'm removing it for now. It was a poor experimental impl [... cut]
closed 2024-12-15 by jamwaffles