ethercrab-rs/ethercrab #265: Compile failure for ek1100 example on master

fpdotmonkey, 2025-01-15:
Hi! Thanks for opening an issue. Please provide some specific details to make it easier to debug any
problems.

## Environment

- Version of ethercrab in use or git hash: 57855f3a28fa281afa075e485a9b0ccf017246d6..0daef104de437bb403f6fedeaa5abf4257c0bd6f (master)
- Operating system kind and version: GNU/Linux 6.6.47+rpt-rpi-2712 (Debian/Raspian)
- EtherCAT devices in use: Beckhoff EK1100, EL3124, EL4134 (though really none)

## Description of the problem/feature request/other

When compiling the library in the above commit range with the below command, I get the following compiler error.
```sh
cargo build --release --example ek1100
```
```
error[E0599]: no variant or associated item named `PreciseCapture` found for enum `TypeParamBound` in the current scope
   --> /home/hydraulics/.cargo/registry/src/index.crates.io-6f17d22bba15001f/savefile-derive-0.18.5/src/lib.rs:321:29
    |
321 |             TypeParamBound::PreciseCapture(c) => {
    |                             ^^^^^^^^^^^^^^ variant or associated item not found in `TypeParamBound`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `savefile-derive` (lib) due to 1 previous error
```

Seemingly a failure in a dependency, but possibly caused by some proc macro shenanigans?

jamwaffles, 2025-01-15:
Thanks for the details! Can I ask what version of Rust you're using? And is it installed from rustup or elsewhere?

fpdotmonkey, 2025-01-15:
It appears that in different commits in that range, I'm using 1.79 or 1.80 according to /rust-toolchain.toml.  I get rust from rustup.

fpdotmonkey, 2025-01-15:
Ah and the build failure *doesn't* occur just building the library.  It's specifically if you're building ek1100.

jamwaffles, 2025-01-15:
I don't have a Raspberry Pi set up to test this on at the moment, but a few months ago I did have everything building on both a Pi 4 and Pi 5. I can also build the examples fine on my AMD dev machine with

```bash
cargo build --release --example ek1100 --target aarch64-unknown-linux-gnu
```

although I do get link errors for the older 32 bit targets when trying `cargo build --release --example ek1100 --target armv7-unknown-linux-gnueabihf`

Have you tried all the usual `cargo clean`, `rm -rf ~/.cargo/registry`, etc commands? You might also want to try bumping the Rust version to `stable` in `rust-toolchain.toml`.

fpdotmonkey, 2025-01-15:
I tried all three of those, including removing rust-toolchain.toml, target/, and ~/.cargo/registry/ entirely, but still no dice.  Doesn't appear to be any problematic build state.  I also did a system update to be sure, and same result (though I didn't reboot, but I doubt that the build is failing due to an older kernel).

fpdotmonkey, 2025-01-15:
```
table-aarch64-unknown-linux-gnu (default)
```

jamwaffles, 2025-01-15:
Thanks. I reproduced on my Pi 4:

```bash
pi@pi4:~/Repositories/ethercrab $ rustc --version
rustc 1.80.1 (3f5fd8dd4 2024-08-06)

pi@pi4:~/Repositories/ethercrab $ uname -a
Linux pi4 6.6.40-rt36-v8+ #3 SMP PREEMPT_RT Sun Jul 21 13:31:46 BST 2024 aarch64 GNU/Linux

EtherCrab git commit:
0daef104de437bb403f6fedeaa5abf4257c0bd6f

pi@pi4:~/Repositories/ethercrab $ rustup default
stable-aarch64-unknown-linux-gnu (default)

pi@pi4:~/Repositories/ethercrab $ rm -rf ~/.cargo/registry/
pi@pi4:~/Repositories/ethercrab $ rm -rf ~/.cargo/.package-cache
pi@pi4:~/Repositories/ethercrab $ rm -rf ~/.cargo/.package-cache-mutate
pi@pi4:~/Repositories/ethercrab $ rm -rf ~/.cargo/.global-cache
pi@pi4:~/Repositories/ethercrab $ cargo clean

cargo clean && cargo build --example ek1100 --release
```

This is not reproducible as a cross-compile which is frustrating.

jamwaffles, 2025-01-15:
Can you try a `cargo update` and see if that helps? I did that on my Pi 4 just now and the example compiles successfully. `TypeParamBound::PreciseCapture` was added in [syn 2.0.81](https://github.com/dtolnay/syn/releases/tag/2.0.81) so make sure Cargo resolves at least that version (e.g. `cargo tree -p syn`, etc).

fpdotmonkey, 2025-01-15:
Ok yeah that did it

jamwaffles, 2025-01-15:
Awesome! I'll close this issue as resolved then :)

closed 2025-01-15 by jamwaffles