ethercrab-rs/ethercrab #286: Waiting for Subdevice to enter init

NothinRandom, 2025-03-04:
## Environment

- Version of ethercrab in use or git hash: [0.5.3]
- Operating system kind and version:
    - [MacOS] [14.6.1] (aarch64)
    - [Ubuntu] [24.04] (x86)
- EtherCAT devices in use:
    - MurrElecktronik 56527: https://shop.murrelektronik.com/en/CUBE67-BUS-NODE-56527.html

## Uploads
N/A

## Description of the problem/feature request/other
When running in MacOS or Ubuntu, I'm keep getting stuck on init.  On Mac it fails out after a few seconds, but on Linux it hangs indefinitely.  I'm not sure what I'm doing wrong here.
### MacOS
```
[2025-03-03 23:41:11.817] [INFO] Discovering EtherCAT devices on en6...
[2025-03-03 23:41:11.818] [DEBUG] Setting source MAC to 80-6d-97-25-74-ad
[2025-03-03 23:41:11.818] [DEBUG] Opening en6 with MTU 4096
[2025-03-03 23:41:11.819] [DEBUG] Discovered 1 SubDevices
[2025-03-03 23:41:11.819] [DEBUG] Beginning reset
[2025-03-03 23:41:11.825] [DEBUG] --> Reset complete
[2025-03-03 23:41:11.825] [DEBUG] Waiting for SubDevice 0x1000 to enter Init
[2025-03-03 23:41:13.541] [DEBUG] Dropping in-flight future, possibly caused by timeout

thread 'main' panicked at src/main.rs:271:10:
Init: Timeout
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
### Linux
```
[2025-03-04 07:33:09.463] [INFO] Discovering EtherCAT devices on enp3s0...
[2025-03-04 07:33:09.544] [DEBUG] Opening enp3s0 with MTU 1500
[2025-03-04 07:33:09.545] [DEBUG] Discovered 1 SubDevices
[2025-03-04 07:33:09.545] [DEBUG] Beginning reset
[2025-03-04 07:33:09.548] [DEBUG] --> Reset complete
[2025-03-04 07:33:09.548] [DEBUG] Waiting for SubDevice 0x1000 to enter Init
```

## Code used to produce the bug
Using the example in README up to `let mut tick_interval = tokio::time::interval(Duration::from_millis(5));`

jamwaffles, 2025-03-04:
Thanks for the report! Could you please attach a Wireshark capture and run with `RUST_LOG=ethercrab=debug`? It's difficult to tell what's happening without a bit more info.

NothinRandom, 2025-03-05:
Thanks @jamwaffles,

This is output from running `RUST_LOG=ethercrab=debug cargo run`
```
dev1:

[ethercat_test_20250304_.pcapng.gz](https://github.com/user-attachments/files/19084036/ethercat_test_20250304_.pcapng.gz)

 admin$ RUST_LOG=ethercrab=debug cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.28s
     Running `target/debug/ethercat_test`

thread 'main' panicked at src/main.rs:480:10:
Init: Timeout
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I gzipped the pcap since upload didn't allow.  Please let me know if there's anything else you need.  Thanks for your help!

jamwaffles, 2025-03-05:
Thanks for the capture! In that Wireshark trace, it's timing out when receiving a response from an EEPROM operation. You can try increasing [`Timeouts::eeprom`](https://docs.rs/ethercrab/latest/ethercrab/struct.Timeouts.html#structfield.eeprom) to something like 30ms (default 10), but this is a bit of a bandaid as you'll likely get timeouts in other places due to hangs and stalls in the USB NIC software stack. Do you have a different USB NIC you can try, or a non-USB one available?

As for Linux, I'm guessing the hang/timeout is also due to poor network performance, but this is a much easier problem to solve. You can use `ethtool` to disable interrupt coalescing which looks like this for Intel (and most other) NICs:

```bash
sudo ethtool -C [nic name here] tx-usecs 0 rx-usecs 0
```

You should also make sure you're using a kernel with the PREEMPT_RT patches enabled.

Finally, if you're using `tokio`, I strongly recommend using `smol` as an executor instead, at least for the EtherCrab tasks. In my testing, `tokio` performs much worse than `smol`.

NothinRandom, 2025-03-06:
Hmm.  I tried multiple timeout settings and different USB Ethernet, but still same issue.  I tried this on a Linux machine with dedicated Ethernet port instead of USB Ethernet, but same issue with hanging.  Just to make sure hardware works, I tried another library called Etherage and was able to identify the device immediately.  For me, I like EtherCrab more because it can compile and work on my Mac laptop, whereas the Etherage only works on Windows/Linux.

I've attached pcap for both and noticed that source mac address using Etherage is broadcast (e.g. `ff:ff:ff:ff:ff:ff`) whereas Ethercrab uses `12:10:10:10:10:10`.  So maybe there is a setting that I'm not using correctly?

[etherage_20250305_1800.pcap.gz](https://github.com/user-attachments/files/19100015/etherage_20250305_1800.pcap.gz)
[ethercrab_20250305_1800.pcap.gz](https://github.com/user-attachments/files/19100016/ethercrab_20250305_1800.pcap.gz)

### Ethercrab code
```
// EtherCrab discover
use env_logger::Env;
use ethercrab::{
    error::Error, std::{ethercat_now, tx_rx_task}, MainDevice, MainDeviceConfig, PduStorage, Timeouts
};
use std::{sync::Arc, time::Duration};
use tokio::time::MissedTickBehavior;

/// Maximum number of SubDevices that can be stored. This must be a power of 2 greater than 1.
const MAX_SUBDEVICES: usize = 16;
/// Maximum PDU data payload size - set this to the max PDI size or higher.
const MAX_PDU_DATA: usize = 1100;
/// Maximum number of EtherCAT frames that can be in flight at any one time.
const MAX_FRAMES: usize = 16;
/// Maximum total PDI length.
const PDI_LEN: usize = 64;

static PDU_STORAGE: PduStorage<MAX_FRAMES, MAX_PDU_DATA> = PduStorage::new();

#[tokio::main]
async fn main() -> Result<(), Error> {
    env_logger::Builder::from_env(Env::default().default_filter_or("debug")).init();

    // let interface = std::env::args()
    //     .nth(1)
    //     .expect("Provide network interface as first argument.");
    let interface: String = "enp3s0".to_string();

    // log2::info!("Starting EK1100 demo...");
    // log2::info!("Ensure an EK1100 is the first SubDevice, with any number of modules connected after");
    log2::info!("Run with RUST_LOG=ethercrab=debug or =trace for debug information");

    let (tx, rx, pdu_loop) = PDU_STORAGE.try_split().expect("can only split once");

    let maindevice = Arc::new(MainDevice::new(
        pdu_loop,
        Timeouts {
            state_transition: Duration::from_millis(5000),
            pdu: Duration::from_micros(30_000),
            eeprom: Duration::from_millis(50),
            wait_loop_delay: Duration::from_millis(2),
            mailbox_echo: Duration::from_millis(100),
            mailbox_response: Duration::from_millis(1000),
        },
        MainDeviceConfig::default(),
    ));

    tokio::spawn(tx_rx_task(&interface, tx, rx).expect("spawn TX/RX task"));

    let mut group = maindevice
        .init_single_group::<MAX_SUBDEVICES, PDI_LEN>(ethercat_now)
        .await
        .expect("Init");

    log2::info!("Discovered {} SubDevices", group.len());

    // for subdevice in group.iter(&maindevice) {
    //     // Special case: if an EL3004 module is discovered, it needs some specific config during
    //     // init to function properly
    //     if subdevice.name() == "EL3004" {
    //         log2::info!("Found EL3004. Configuring...");
[... 96 lines cut ...]
                    i,
                    can.sdo_read_slice(&sdo::device::name, priority, &mut name).await
                        .map(|s|  std::str::from_utf8(s).unwrap().trim_end())
                        .unwrap_or("unidentified"),
                    info.ty(),
                    info.revision(),
                    info.build(),
                    can.sdo_read_slice(&sdo::device::hardware_version, priority, &mut hardware).await
                        .map(|s| std::str::from_utf8(s).unwrap().trim_end())
                        .unwrap_or("NA"),
                    can.sdo_read_slice(&sdo::device::software_version, priority, &mut software).await
                        .map(|s| std::str::from_utf8(s).unwrap().trim_end())
                        .unwrap_or("NA"),
                    );
            Result::<_, Box<dyn Error>>::Ok(())
        };
        if let Err(err) = task.await
            {println!("slave {}: failure: {:?}", i, err)}
    }

    Ok(())
}
```

Outputs: `slave 0: Cube67+ BN-EC`

jamwaffles, 2025-03-19:
Thanks for the fixed Wireshark trace.

1. The final frame sent before timing out has a ~350ms delay from the previous received frame, which is odd because it's in the middle of an EEPROM transaction. You could try increasing the EEPROM timeout to something silly like 1000ms and see if that helps anything, but if you make it to OP and these hangs are caused by something in the OS/NIC/who knows, you'll of course hit other issues during the process data cycle.
2. I'm not sure why it's panicking with `attempt to add with overflow`, but regardless of whether that's a true bug or not, it implies you're running _without_ the `--release` flag. Is this correct?

I'll see if I can figure out why it's panicking with an overflow when I have a bit more time/energy.

jamwaffles, 2025-03-20:
So, I did a bit of digging and opened a fix in #288. I'll release that as a 0.5.x version in a bit so you can `cargo update`. I believe this will fix your issue with both that panic message and infinite hangs, however to be absolutely certain would you mind uploading a dump of the SubDevice's EEPROM?

You can run the `dump-eeprom` example in this repo, e.g. 

```bash
# SubDevice is assumed to be first in network with `0` index:
cargo run --example dump-eeprom [NIC here] 0 > murr-56527.bin
```

If not, no worries, but it would be helpful to have the dump so I can add a regression test :)

jamwaffles, 2025-03-20:
EtherCrab 0.5.4 is out - please `cargo update` and test when you have a moment.

NothinRandom, 2025-03-20:
@jamwaffles,

Nice work!  We are making progress when updating to 0.5.4.  Now I get the following output:
```
[2025-03-20 15:12:02.985] [INFO] Run with RUST_LOG=ethercrab=debug or =trace for debug information
[2025-03-20 15:12:02.986] [DEBUG] Setting source MAC to 00-e0-4c-68-6b-b6
[2025-03-20 15:12:02.986] [DEBUG] Opening en20 with MTU 4096
[2025-03-20 15:12:02.987] [DEBUG] Discovered 1 SubDevices
[2025-03-20 15:12:02.987] [DEBUG] Beginning reset
[2025-03-20 15:12:02.992] [DEBUG] --> Reset complete
[2025-03-20 15:12:02.992] [DEBUG] Waiting for SubDevice 0x1000 to enter Init
[2025-03-20 15:12:03.036] [DEBUG] Did not find any non-empty categories. EEPROM could be empty or corrupt.
[2025-03-20 15:12:03.037] [DEBUG] SubDevice 0x1000 name manu. 0x0000004f, device 0x0000dccf, serial 0x00000000 vendor: 0x0000004f, product 0x0000dccf, rev 1, serial 0, DC: no, ports [ open closed closed closed ], alias address 0x0000
[2025-03-20 15:12:03.037] [DEBUG] Configuring topology/distributed clocks
[2025-03-20 15:12:03.037] [DEBUG] SubDevice 0x1000 manu. 0x0000004f, device 0x0000dccf, serial 0x00000000 DC: no
[2025-03-20 15:12:03.037] [DEBUG] No SubDevices with DC support found
[2025-03-20 15:12:03.037] [DEBUG] Distributed clock config complete
[2025-03-20 15:12:03.037] [DEBUG] Going to configure group with 1 SubDevice(s), starting PDI offset 0x000000
[2025-03-20 15:12:03.076] [DEBUG] Did not find any non-empty categories. EEPROM could be empty or corrupt.
[2025-03-20 15:12:03.076] [DEBUG] Discovered sync managers:
[]
[2025-03-20 15:12:03.112] [DEBUG] Did not find any non-empty categories. EEPROM could be empty or corrupt.

thread 'main' panicked at src/main.rs:620:10:
Init: Eeprom(NoCategory)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Changing eeprom timeout from 200ms to 1000ms still yields panic.  Same when running in release mode.  I can change eeprom timeout back to 30ms, and it works fine... up to panic.

It's currently stuck on this line:
```
let mut group = maindevice
        .init_single_group::<MAX_SUBDEVICES, PDI_LEN>(ethercat_now)
        .await
        .expect("Init");
```

jamwaffles, 2025-03-21:
Thanks for testing and my bad that the fix didn't work! I don't have a device with an empty EEPROM so I wasn't able to test end to end, although thinking about it now I could use a mock EEPROM file... Either way, let me do some more involved debugging and I'll try and get you a more reliable fix. Off the top of my head I don't think the `init` methods need much from the EEPROM so should be failing gracefully instead of erroring.

jamwaffles, 2025-03-21:
I just published some more fixes from #291 in v0.5.5 - please `cargo update` again when you can.

In my local tests, EtherCrab now gets past `init()` with an empty EEPROM, however I'm not sure what else is missing to get the SubDevice into OP and I have a strong suspicion you'll encounter more errors. 

The most common way to configure EtherCAT devices is with ESI files, which I'm guessing this SubDevice manufacturer is relying on and not bothering to populate the EEPROM. EtherCrab doesn't currently support ESI files, however I'm making some slow steps towards that functionality so if you do indeed encounter further issues, watch this space...

NothinRandom, 2025-03-21:
@jamwaffles,

Good work!  I see progress!

```
[2025-03-21 06:45:57.968] [DEBUG] Setting source MAC to 00-e0-4c-68-6b-b6
[2025-03-21 06:45:57.969] [DEBUG] Opening en20 with MTU 4096
[2025-03-21 06:45:57.969] [DEBUG] Discovered 1 SubDevices
[2025-03-21 06:45:57.969] [DEBUG] Beginning reset
[2025-03-21 06:45:57.979] [DEBUG] --> Reset complete
[2025-03-21 06:45:57.981] [DEBUG] Waiting for SubDevice 0x1000 to enter Init
[2025-03-21 06:45:58.034] [DEBUG] SubDevice 0x1000 name manu. 0x0000004f, device 0x0000dccf, serial 0x00000000 vendor: 0x0000004f, product 0x0000dccf, rev 1, serial 0, DC: no, ports [ open closed closed closed ], alias address 0x0000
[2025-03-21 06:45:58.034] [DEBUG] Configuring topology/distributed clocks
[2025-03-21 06:45:58.034] [DEBUG] SubDevice 0x1000 manu. 0x0000004f, device 0x0000dccf, serial 0x00000000 DC: no
[2025-03-21 06:45:58.035] [DEBUG] No SubDevices with DC support found
[2025-03-21 06:45:58.035] [DEBUG] Distributed clock config complete
[2025-03-21 06:45:58.035] [DEBUG] Going to configure group with 1 SubDevice(s), starting PDI offset 0x000000
[2025-03-21 06:45:58.065] [DEBUG] Discovered sync managers:
[]
[2025-03-21 06:45:58.094] [DEBUG] 0x1000 has no EEPROM general category, using default
[2025-03-21 06:45:58.095] [DEBUG] SubDevice 0x1000 mailbox SMs configured. Transitioning to PRE-OP
[2025-03-21 06:45:58.095] [DEBUG] Set state Pre-Operational for SubDevice address 0x1000
```

I am excited to see ESI parsing incorporated; can't wait to test out next release.

jamwaffles, 2025-07-05:
Going to close this as I think the original issue is resolved. Please reopen or file a new issue if there's something else still not working :)

closed 2025-07-05 by jamwaffles