ethercrab-rs/ethercrab #293: Failed to claim receiving frame 13: expected state `Sent`, but got `None` (or `RxProcessing`, or `Sendable`)

1cedsoda, 2025-03-21:
Hi, when I run the ek1100 example, the tx_rx_taks fails after a while. Sometimes after a couple seconds, sometime after some minutes.
This is a reproducible example, but I got this and similar errors in my own codebase.
Therese errors were not introduced very recently.

## Environment

- Version of ethercrab in use or git hash: 7fbd959c62c31b179d2c54259aee3071d761f162
- Operating system kind and version: OSX 15.3.1
- EtherCAT devices in use: EK1100, EL2002, EL3001, EL2521, EL2522

## Uploads
### Log using ek1100 example:
`RUST_LOG=debug cargo run --example ek1100 en6`

recieved `None`
```
[2025-03-21T14:18:17Z INFO  ek1100] -> SubDevice 0x1000 EK1100 inputs: 0 bytes, outputs: 0 bytes
[2025-03-21T14:18:17Z INFO  ek1100] -> SubDevice 0x1001 EL2002 inputs: 0 bytes, outputs: 1 bytes
[2025-03-21T14:18:17Z INFO  ek1100] -> SubDevice 0x1002 EL3001 inputs: 2 bytes, outputs: 0 bytes
[2025-03-21T14:18:17Z INFO  ek1100] -> SubDevice 0x1003 EL2521-0024 inputs: 8 bytes, outputs: 14 bytes
[2025-03-21T14:18:17Z INFO  ek1100] -> SubDevice 0x1004 EL2522 inputs: 16 bytes, outputs: 28 bytes
[2025-03-21T14:18:17Z INFO  ek1100] -> SubDevice 0x1005 EL2522 inputs: 16 bytes, outputs: 28 bytes
thread 'main' panicked at examples/ek1100.rs:122:40:
TX/RX: Timeout
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[2025-03-21T14:19:56Z ERROR ethercrab::pdu_loop::frame_element] Failed to claim receiving frame 4: expected state Sent, but got None
[2025-03-21T14:19:56Z ERROR ethercrab::std::unix] Failed to receive frame: pdu: invalid PDU index 4
```

### Similar errors I encountered with my codebase:
recieved`RxProcessing`
```
[2025-03-21T12:09:45Z ERROR ethercrab::pdu_loop::frame_element] Failed to claim receiving frame 13: expected state Sent, but got RxProcessing
[2025-03-21T12:09:45Z ERROR ethercrab::std::unix] Failed to receive frame: pdu: invalid PDU index 13
```

recieved`Sendable`
```
[2025-03-21T12:16:18Z ERROR ethercrab::pdu_loop::frame_element] Failed to claim receiving frame 10: expected state Sent, but got Sendable
[2025-03-21T12:16:18Z ERROR ethercrab::std::unix] Failed to receive frame: pdu: invalid PDU index 10
```
### And this one
```
[2025-03-21T12:11:43Z ERROR ethercrab::std::unix] Failed to receive frame: pdu: failed to decode raw PDU data into type
```

jamwaffles, 2025-03-21:
Hey, thanks for the report. For spurious issues like this, please always upload a Wireshark trace if you can capture the failing event. Having the packet logs makes debugging a bit easier on my side :).

Are you able to test the same example on a Linux machine, and if so do you see the same result? I haven't properly tested the macOS network driver in quite some time, so there could be a bug buried in it somewhere.

jamwaffles, 2025-03-21:
Just had a quick look and I see a 57ms hang between the penultimate received and the final sent frame. Not sure if the "failed to claim receiving frame" is a cause or a symptom of that but I'll look a bit deeper tomorrow maybe.

In the meantime, you could try bumping [this config value](https://docs.rs/ethercrab/latest/ethercrab/struct.Timeouts.html#structfield.pdu) and see where you get although it kinda smells like there's some weirdness going on in macOS.

By the way, you can add a column called "delta time" in Wireshark that will show network round trip times and hangs between loop iterations. I can't remember if I mentioned this before, but if you're using `tokio`, I'd recommend switching to the `smol` executor (and timers, etc) instead as `tokio` can have issues with latency spikes.

andergisomon, 2025-03-28:
Had a similar issue. I ended up just blindly increasing the timeouts in maindeviceconfigs. Even then it still takes a couple tries to run the script for it to start detecting the subdevices and running.

jamwaffles, 2025-03-28:
I can also reproduce this occasionally on an M1 MacBook Air using a USB-C ethernet adapter, running with `cargo run --example ek1100 en11` (NOT with `--release`). I'm checked out to tag `ethercrab-v0.5.6`.

The root cause will be a race condition in the `unsafe` (🤫) internals of EtherCrab which will be fun to track down...

jamwaffles, 2025-03-28:
I noticed that when my Mac (plugged in) went to screensaver, the failed to claim error occurred and the example program stopped. This led me to a method to somewhat reliably reproduce the failure:

- Change tick rate in `ek1100` example to `1ms`
- `cargo run --example ek1100 en11` (optionally with `--release`)
- Type in `screensaver` in Spotlight and open the settings window for it
- The EK1100 example will sometimes crash with `Failed to claim receiving frame ...`, sometimes with a non-logging timeout panic, or sometimes keep working fine

All this said, the error itself is a bit of a red herring. When a timeout occurs, some internal state is cleaned up by custom `Drop` impls, so when a response frame is received and the TX/RX task is still running it will try to fetch the now-invalid slot that frame belongs to, causing the error.

@1cedsoda For the other errors you posted under `Similar errors I encountered with my codebase:`, do they all occur after the timeout panic, or before it?

I have a few things to try:

1. Always run with `--release` - this mitigates the crashyness a bit
2. Consider increasing cycle times and when running on macOS with `#[cfg()]` attributes
3. Do the same with [`Timeouts::pdu`](https://docs.rs/ethercrab/latest/ethercrab/struct.Timeouts.html#structfield.pdu)
4. Turn on TX retries with [`MainDeviceConfig::retry_behaviour`](https://docs.rs/ethercrab/latest/ethercrab/struct.MainDeviceConfig.html#structfield.retry_behaviour)
5. Use the `smol` executor instead of `tokio`. This is general advice for anyone seeking better EtherCrab performance really. `tokio` isn't as consistent as `smol` and seems to suffer when system load is higher.

If the errors are always _after_ a timeout panic, I'm afraid there's no definitive bug to fix or action to take, so I'll probably close this issue. Let me know how you get on.

1cedsoda, 2025-04-03:
Can't see any timeout panic here
```
[2025-04-03T13:48:00Z INFO  server::ethercat::r#loop] Starting control loop
[2025-04-03T13:50:13Z ERROR ethercrab::pdu_loop::frame_element] Failed to claim receiving frame 10: expected state Sent, but got Sendable
[2025-04-03T13:50:13Z ERROR ethercrab::std::unix] Failed to receive frame: pdu: invalid PDU index 10
```

I will try the things you proposed

jamwaffles, 2025-04-03:
Hm. Strange. Are you on v0.6? That might have something magical in it that fixes everything.

As usual, please attach Wireshark captures, full logs and source code if it's easy enough to capture these failures. I can't do much without the full picture. At minimum, the Wireshark capture will tell me if everything is ok on the network side.

Are you sending a lot of traffic with a fast cycle time? If you are, consider increasing the `MAX_FRAMES` const generic, i.e. [this one](https://github.com/ethercrab-rs/ethercrab/blob/main/examples/ek1100.rs#L35).

andergisomon, 2025-04-07:
Quick update: Problem seems to have disappeared with 0.6.0 (at least for me). Although this time I ran the ek1100 example on its own (not a clone repo). 

[ek1100_capture_0.6.0_2025-04-07T09:13+08:00.zip](https://github.com/user-attachments/files/19624538/ek1100_capture_0.6.0_2025-04-07T09.13%2B08.00.zip)

jamwaffles, 2025-07-05:
@1cedsoda I'll close this as resolved unless you're still experiencing the issue. If so, please reopen.

Oshgnacknak, 2025-08-07:
Hi, @1cedsoda, me and a few others from our group are still struggling with disconnects on our machines. 
One machine contains roughly 10 sub devices all connected with two EK1100 connected in serial.

Ethercat connections are managed by ethercrab now running on a real time thread, as suggested. This helped for a few weeks, however, disconnects keep reappearing.
We assume that having more devices makes disconnects more likely.
You can see how we initialize ethercrab here: https://github.com/qitechgmbh/control/blob/57f6e50e0e7eadd43c8cf39e855e30fa72a98108/server/src/ethercat/setup.rs#L43

We caught two disconnects with wireshark: https://mega.nz/file/DR0gXCLD#fnOkjBQ1BEN6_8Zq5hflbtXEsaYu-vvQ3rzxVIJerp4
In this share, you can also see our log files. We assume that the most important lines are
```
Aug 05 14:11:24 nixos server[995]: Failed to receive frame: pdu: failed to decode raw PDU data into type
Aug 05 14:11:24 nixos server[995]: Failed to receive frame: pdu: failed to decode raw PDU data into type
Aug 05 14:11:25 nixos server[995]: Memory: 2_986 KB (-9 KB/s)
Aug 05 14:11:25 nixos server[995]: Memory: 2_986 KB (-9 KB/s)
Aug 05 14:11:29 nixos server[995]: Loop failed
                                   timeout
                                   Stack backtrace:
                                      0: anyhow::error::<impl core::convert::From<E> for anyhow::Error>::from
                                      1: <tracing::instrument::Instrumented<T> as core::future::future::Future>::poll
                                      2: server::loop::init_loop::{{closure}}::{{closure}}
                                      3: <futures_lite::future::Or<F1,F2> as core::future::future::Future>::poll
                                      4: std::thread::local::LocalKey<T>::with
                                      5: async_io::driver::block_on
                                      6: server::loop::init_loop::{{closure}}
                                      7: std::sys::backtrace::__rust_begin_short_backtrace
                                      8: std::thread::Builder::spawn_unchecked_::{{closure}}::{{closure}}
                                      9: <core::panic::unwind_safe::AssertUnwindSafe<F> as core::ops::function::FnOnce<()>>::call_once
                                     10: core::ops::function::FnOnce::call_once{{vtable.shim}}
                                     11: std::sys::pal::unix::thread::Thread::new::thread_start
                                     12: start_thread
                                     13: clone
Aug 05 14:11:29 nixos systemd[1]: qitech-control-server.service: Main process exited, code=exited, status=1/FAILURE
Aug 05 14:11:29 nixos systemd[1]: qitech-control-server.service: Failed with result 'exit-code'.
```
and
```
Aug 05 15:02:06 nixos server[999]: Failed to claim receiving frame 13: expected state Sent, but got RxDone
Aug 05 15:02:06 nixos server[999]: Failed to receive frame: pdu: invalid PDU index 13
Aug 05 15:02:06 nixos server[999]: Failed to claim receiving frame 13: expected state Sent, but got RxDone
Aug 05 15:02:06 nixos server[999]: Failed to receive frame: pdu: invalid PDU index 13
Aug 05 15:02:11 nixos server[999]: Loop failed
                                   timeout
                                   Stack backtrace:
                                      0: anyhow::error::<impl core::convert::From<E> for anyhow::Error>::from
                                      1: <tracing::instrument::Instrumented<T> as core::future::future::Future>::poll
                                      2: server::loop::init_loop::{{closure}}::{{closure}}
                                      3: <futures_lite::future::Or<F1,F2> as core::future::future::Future>::poll
                                      4: std::thread::local::LocalKey<T>::with
                                      5: async_io::driver::block_on
                                      6: server::loop::init_loop::{{closure}}
                                      7: std::sys::backtrace::__rust_begin_short_backtrace
                                      8: std::thread::Builder::spawn_unchecked_::{{closure}}::{{closure}}
                                      9: <core::panic::unwind_safe::AssertUnwindSafe<F> as core::ops::function::FnOnce<()>>::call_once
                                     10: core::ops::function::FnOnce::call_once{{vtable.shim}}
                                     11: std::sys::pal::unix::thread::Thread::new::thread_start
                                     12: start_thread
                                     13: clone
Aug 05 15:02:11 nixos systemd[1]: qitech-control-server.service: Main process exited, code=exited, status=1/FAILURE
Aug 05 15:02:11 nixos systemd[1]: qitech-control-server.service: Failed with result 'exit-code'.
```

## Environment

Rust Version: 1.86 
ethercrab version: 0.6

jamwaffles, 2025-08-11:
@Oshgnacknak

Both `crash1/err.pcapng` and `crash2/crash.pcapng` show semi regular RTT spikes of up to 1ms. Your cycle time is also 1ms, which leaves no time left for any processing.

[image]

Have you configured your NIC with `ethtool`? I can't remember how but you can definitely do it with Nix. `tx-usecs` and `rx-usecs` should be `0`. You can check the current value with `ethtool -c <interface>`.

The cycle-cycle delta is also very odd. It flips between ~800us and ~100us:

[image]

This value should be consistent and very close to your configured cycle time. I think `LoopThrottle` might not be doing what you want it to do. Try increasing the cycle time to 2ms and see if anything gets better. I don't think you need 1ms but I'll leave that up to you.

This could possibly be the reason that packet timeouts occur - you might be sending stuff too fast for the SubDevices to respond. Beckhoff specifies 125us minimum cycle time for some/all their devices and a 100us cycle-cycle delta is under that.

You might want to switch to the blocking [`tx_rx_io_uring`](https://docs.rs/ethercrab/latest/ethercrab/std/fn.tx_rx_task_io_uring.html) for a bit more performance.

I'd also recommend enabling [Distributed Clocks](https://github.com/ethercrab-rs/ethercrab/blob/main/examples/dc.rs) and using the EtherCAT system time on the EK1100 as the cycle reference. You're still somewhat at the mercy of Linux timers but using something like [`timerfd`](https://docs.rs/timerfd/latest/timerfd/) can give good results. DC is a lot more involved to set up though.